nasa/fprime #1208: Unit Test code coverage fails

kevin-f-ortega, 2022-01-06:
| | |
|:---|:---|
|**_F´ Version_**| 65d960f11a39a4463c6396318602243440cd5846|
|**_Affected Component_**|  All |
---
## Problem Description

Running `fprime-util check --coverage` fails with the following error
```
100% tests passed, 0 tests failed out of 1

Total Test time (real) =   0.01 sec
[100%] Built target Abc_Svc_Time_check
CMakeFiles/Abc_Svc_Time.dir/Time.gcno:cannot open notes file
CMakeFiles/Abc_Svc_Time.dir/Time.gcda:cannot open data file, assuming not executed
CMakeFiles/Abc_Svc_Time.dir/TimeComponentAc.gcno:cannot open notes file
CMakeFiles/Abc_Svc_Time.dir/TimeComponentAc.gcda:cannot open data file, assuming not executed
'/home/ortega/abc/Abc/build-fprime-automatic-native-ut/Abc/Svc/Time/TimeComponentAc.cpp' file is already processed
Error copying file "*.gcov" to "/home/ortega/abc/Abc/Svc/Time/coverage".
make[3]: *** [Abc/Svc/Time/CMakeFiles/Abc_Svc_Time_coverage.dir/build.make:72: Abc/Svc/Time/CMakeFiles/Abc_Svc_Time_coverage] Error 1
make[2]: *** [CMakeFiles/Makefile2:33092: Abc/Svc/Time/CMakeFiles/Abc_Svc_Time_coverage.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:33099: Abc/Svc/Time/CMakeFiles/Abc_Svc_Time_coverage.dir/rule] Error 2
make: *** [Makefile:12491: Abc_Svc_Time_coverage] Error 2
[ERROR] CMake erred with return code 2

```
## How to Reproduce

1. cd into a component's directory
2. Run `fprime-util check --coverage`

## Expected Behavior

I expected gcov files in my component's directory.

LeStarch, 2022-01-06:
As part of this, we should automatically post our coverage somewhere such that we don't introduce regressions.

Joshua-Anderson, 2022-01-06:
Take a look at codecov https://about.codecov.io/ - it should be pretty easy to get setup with github actions and our gcov coverage format. It can automatically notify about code coverage regressions in PRs.

ThibFrgsGmz, 2022-01-07:
@Joshua-Anderson Ah perfect, since a few weeks I was preparing a message for the Discussions where I asked if there was no way to add a badge on the README.md of the project where we display the percentage of code coverage.

This is a question that comes up a lot when lobbying in my company and people would be more willing to use an open source framework if for example it had 100% MC/DC coverage to develop Category A projects more easily.

Codecov was my recommendation since we see it a lot on open source projects.
But I wonder if there is not something better, something free even for private repo.

ThibFrgsGmz, 2022-01-07:
The goal of having a Category A certifiable framework, 100% MC/DC coverage, and the full test suite is a powerful selling point.
A selling point that cFS will soon have, handled by NASA-JSC if I remember correctly, since it will be "mandatory" for Artemis.

LeStarch, 2022-01-11:
Confirmed issue!

kevin-f-ortega, 2022-01-26:
Addressed in https://github.com/nasa/fprime/pull/1237

closed 2022-01-26 by kevin-f-ortega