nasa/fprime #1217: Enum type in nested namespace generates malformed include guard

vietjtnguyen, 2022-01-08:
| | |
|:---|:---|
|**_F´ Version_**| v3.0.0 |
|**_Affected Component_**| Autocoder |
---
## Problem Description

Nesting an enum type in FPP in more than one module produces a malformed include guard in the autocode output. It also produces a nested namespace definition which is valid only in the C++17 standard and later.

## How to Reproduce

```
module Cadre {
module Fprime {
  enum MotorNames : I32 {
    FRONT_LEFT = 0,
    FRONT_RIGHT = 1,
    BACK_LEFT = 2,
    BACK_RIGHT = 3
  } default FRONT_LEFT
}
}
```

Produces...

```cpp
...
#ifndef Cadre::Fprime_MotorNames_HPP
#define Cadre::Fprime_MotorNames_HPP

#include "Fw/Types/String.hpp"
#include "Fw/Types/BasicTypes.hpp"
#include "Fw/Types/Serializable.hpp"

namespace Cadre::Fprime {
  class MotorNames : public Fw::Serializable
  {
...
```

## Expected Behavior

```cpp
...
#ifndef CADRE_FPRIME_MOTORNAMES_HPP_
#define CADRE_FPRIME_MOTORNAMES_HPP_

#include "Fw/Types/String.hpp"
#include "Fw/Types/BasicTypes.hpp"
#include "Fw/Types/Serializable.hpp"

namespace Cadre {
namespace Fprime {
  class MotorNames : public Fw::Serializable
  {
...
```

LeStarch, 2022-09-26:
@bocchino thoughts here?

bocchino, 2022-09-26:
This issue will go away when we merge https://github.com/nasa/fprime/pull/1630.

LeStarch, 2022-11-30:
The fixing PR was merged.

closed 2022-11-30 by LeStarch