nasa/fprime #1218: Array types in autocoded serializable classes have identity-equality instead of value-equality comparison operators

vietjtnguyen, 2022-01-10:
| | |
|:---|:---|
|**_F´ Version_**| v3.0.0 j
|**_Affected Component_**| autocoded serializables with arrays |
---
## Problem Description

Looking at the autocode for equality comparison for autocoded serializable types...

https://github.com/nasa/fprime/blob/313ef0556baec4a981f03e66de64d9193a84b867/Autocoders/Python/src/fprime_ac/generators/templates/serialize/publicSerialCpp.tmpl#L54-L56

...we see that for array types it produces a comparison that checks the pointer address to the array which will only return true if the object is compared to itself (identity/entity equality). Here's an example output:

Declaration:

```cpp
class MotorSpeeds : public Fw::Serializable {

public:

    enum {
        SERIALIZED_SIZE =
        sizeof(F32)*4
    }; //!< serializable size of MotorSpeeds

    MotorSpeeds(); //!< Default constructor
    MotorSpeeds(const MotorSpeeds* src); //!< pointer copy constructor
    MotorSpeeds(const MotorSpeeds& src); //!< reference copy constructor
    MotorSpeeds(const F32* angularSpeedRadsPerSecond, NATIVE_INT_TYPE angularSpeedRadsPerSecondSize); //!< constructor with arguments
    MotorSpeeds(const F32 angularSpeedRadsPerSecond); //!< constructor with arguments with scalars for array arguments
    MotorSpeeds& operator=(const MotorSpeeds& src); //!< equal operator
    bool operator==(const MotorSpeeds& src) const; //!< equality operator
#ifdef BUILD_UT
    // to support GoogleTest framework in unit tests
    friend std::ostream& operator<<(std::ostream& os, const MotorSpeeds& obj);
#endif

    void set(const F32* angularSpeedRadsPerSecond, NATIVE_INT_TYPE angularSpeedRadsPerSecondSize); //!< set values

    const F32* getangularSpeedRadsPerSecond(NATIVE_INT_TYPE& size) const; //!< get member angularSpeedRadsPerSecond

    void setangularSpeedRadsPerSecond(const F32* val, NATIVE_INT_TYPE size); //!< set member angularSpeedRadsPerSecond

    Fw::SerializeStatus serialize(Fw::SerializeBufferBase& buffer) const; //!< serialization function
    Fw::SerializeStatus deserialize(Fw::SerializeBufferBase& buffer); //!< deserialization function
#if FW_SERIALIZABLE_TO_STRING || BUILD_UT
    void toString(Fw::StringBase& text) const; //!< generate text from serializable
#endif
protected:

    enum {
        TYPE_ID = 0xA09C7D3A //!< type id
    };

    F32 m_angularSpeedRadsPerSecond[4]; //<! angularSpeedRadsPerSecond - 

private:

};
```

Definition:

```cpp
bool MotorSpeeds::operator==(const MotorSpeeds& src) const {
    return (
        (src.m_angularSpeedRadsPerSecond == this->m_angularSpeedRadsPerSecond) &&
        true);
}
```

I feel that the object is intended to have value-type semantics (copyable, assignable).

LeStarch, 2022-01-10:
Makes sense to me.  Currently we are avoiding updates to the Python Autocoder as FPP is now on the scene....but we should ensure this is included in the FPP autocoder.

LeStarch, 2022-09-26:
@bocchino did we fix this in the array work we did this summer?  If so, will you close this issue?

bocchino, 2022-09-26:
This issue will be be fixed and closed when we merge #1630.

LeStarch, 2022-11-30:
The fixing PR was merged.

closed 2022-11-30 by LeStarch