nasa/fprime #1221: Create reference sub-repo for testing

timcanham, 2022-01-13:
| | |
|:---|:---|
|**_F´ Version_**| V3.0 |
|**_Affected Component_**| all |
---
## Feature Description

Create a reference deployment (probably in https://github.com/fprime-community) that includes fprime as a sub-repo. That could help detect issues like https://github.com/nasa/fprime/pull/1220. This repo would be built and tested for each release of F Prime. Suggestion would be to reproduce the `Ref` app with another name. It could include all the components from the F Prime subrepo.

## Rationale

Help find errors when using F Prime as a subrepo.

LeStarch, 2022-01-13:
We have a Ref repo, it just isn't maintained. We should update it, and add the build to CI.

Joshua-Anderson, 2022-01-14:
Does the GPS app count? https://github.com/fprime-community/gps-tutorial

Joshua-Anderson, 2022-01-14:
We already have testing F' to make sure the gps-tutorial builds

LeStarch, 2022-03-22:
We need to add the `separated-project-example` repo into CI in the same way `gps-tutorial` is done.

LeStarch, 2022-09-26:
Hurray system reference!  **Note:** currently the build is manual, but it exists.

closed 2022-09-26 by LeStarch