nasa/fprime #1229: Add unlock() alias or rename unLock() to unlock() in Os::Mutex to support std::lock_guard

vietjtnguyen, 2022-01-18:
[`std::lock_guard`](https://en.cppreference.com/w/cpp/thread/lock_guard) is a handy type to lock/unlock a mutex automatically according the lifetime of a `std::lock_guard` instance. It only requires that the provided mutex type satisfy [BasicLockable](https://en.cppreference.com/w/cpp/named_req/BasicLockable) which only requires the existence of a `lock()` and `unlock()` member function.

The least intrusive method is to add an alias function:

```diff
diff --git a/Os/Mutex.hpp b/Os/Mutex.hpp
index 94ebab234..31934015a 100644
--- a/Os/Mutex.hpp
+++ b/Os/Mutex.hpp
@@ -14,6 +14,14 @@ namespace Os {
             void lock(); //!<  lock the mutex
             void unLock(); //!<  unlock the mutex

+            /**
+             * An alias to @ref unLock() that allows the type to satisfy the
+             * BasicLockable requirements
+             * (https://en.cppreference.com/w/cpp/named_req/BasicLockable)
+             * needed to work with @ref std::lock_guard.
+             */
+            void unlock() { this->unLock(); }
+
         private:

             POINTER_CAST m_handle; //!<  Stored handle to mutex
```

connorfuhrman, 2022-05-03:
Is there a PR associated with this? I can take this assignment if it's still available.

LeStarch, 2022-05-03:
There is not PR for this.  We should add an alias as opposed to renaming the function.

connorfuhrman, 2022-05-03:
Working at https://github.com/connorfuhrman/fprime/tree/enh/mutex_basiclockable

Will simply add aliases for lock and unlock and write some basic tests.

LeStarch, 2022-06-30:
This was done!

closed 2022-06-30 by LeStarch