nasa/fprime #1238: `fprime-util impl --ut` command not generating Tester.cpp and Tester.hpp in 4.5.1. Set Up the Unit Test Environment

capsulecorplab, 2022-01-26:
| | |
|:---|:---|
|**_F´ Version_**| v3.0.0 |
|**_Affected Component_**| MathSender |
---
## Problem Description

Following steps in [4.5.1. Set Up the Unit Test Environment](https://nasa.github.io/fprime/Tutorials/MathComponent/Tutorial.html#The-MathSender-Component_Write-and-Run-Unit-Tests_Set-Up-the-Unit-Test-Environment), `fprime-util impl --ut` command not generating `Tester.cpp` and `Tester.hpp` files. 

See last commit on forked repo at https://github.com/capsulecorplab/fprime/commit/1c27e65de9d2f83c0e9c9ad236eb59a98386e398 

## How to Reproduce

1. Follow [MathComponent tutorial](https://nasa.github.io/fprime/Tutorials/MathComponent/Tutorial.html) 
2. Run `fprime-util impl --ut` in `Ref/MathSender` directory as per [4.5.1. Set Up the Unit Test Environment](https://nasa.github.io/fprime/Tutorials/MathComponent/Tutorial.html#The-MathSender-Component_Write-and-Run-Unit-Tests_Set-Up-the-Unit-Test-Environment)

## Expected Behavior

`fprime-util impl --ut` should generate `Tester.cpp` and `Tester.hpp` files.

astroesteban, 2022-01-29:
Hey @capsulecorplab I was actually experiencing the same bug. I fixed it as described here: https://github.com/nasa/fprime/discussions/1226#discussioncomment-2070695.
I've got a fix coming out soon.

capsulecorplab, 2022-01-30:
Updated `Ref/MathSender/CMakeLists.txt` in https://github.com/capsulecorplab/fprime/commit/4d888d4e21d1ec2de25ade6158548c0e56c70bac as per https://github.com/nasa/fprime/discussions/1226#discussioncomment-2070695, but ran into the same CMake error, running `fprime-util generate --ut`

```
CMake Error at /home/captainpike/repo/fork/fprime/cmake/target/ut.cmake:72 (set_property):
  set_property could not find TARGET Ref_MathSender.  Perhaps it has not yet
  been created.
Call Stack (most recent call first):
  /home/captainpike/repo/fork/fprime/cmake/target/target.cmake:127 (add_module_target)
  /home/captainpike/repo/fork/fprime/cmake/target/target.cmake:143 (setup_module_target)
  /home/captainpike/repo/fork/fprime/cmake/module.cmake:86 (setup_all_module_targets)
  /home/captainpike/repo/fork/fprime/cmake/API.cmake:427 (generate_ut)
  MathSender/CMakeLists.txt:8 (register_fprime_ut)

CMake Error at /home/captainpike/repo/fork/fprime/cmake/target/coverage.cmake:71 (get_target_property):
-- Adding Deployment: Ref
-- Configuring incomplete, errors occurred!
  get_target_property() called with non-existent target "Ref_MathSender".
See also "/home/captainpike/repo/fork/fprime/Ref/build-fprime-automatic-native-ut/CMakeFiles/CMakeOutput.log".
Call Stack (most recent call first):
See also "/home/captainpike/repo/fork/fprime/Ref/build-fprime-automatic-native-ut/CMakeFiles/CMakeError.log".
  /home/captainpike/repo/fork/fprime/cmake/target/target.cmake:127 (add_module_target)
  /home/captainpike/repo/fork/fprime/cmake/target/target.cmake:143 (setup_module_target)
  /home/captainpike/repo/fork/fprime/cmake/module.cmake:86 (setup_all_module_targets)
  /home/captainpike/repo/fork/fprime/cmake/API.cmake:427 (generate_ut)
  MathSender/CMakeLists.txt:8 (register_fprime_ut)

[ERROR] CMake erred with return code 1. Partial build cache remains. Run purge to clean-up.
```

astroesteban, 2022-01-31:
Hey @capsulecorplab does your `Ref/MathSender/CMakeLists.txt` should look like this:
```cmake
# Register the standard build
set(SOURCE_FILES
  "${CMAKE_CURRENT_LIST_DIR}/MathSender.cpp"
  "${CMAKE_CURRENT_LIST_DIR}/MathSender.fpp"
)
register_fprime_module()

# Register the unit test build
set(UT_SOURCE_FILES
  "${CMAKE_CURRENT_LIST_DIR}/MathSender.fpp"
  "${CMAKE_CURRENT_LIST_DIR}/test/ut/Tester.cpp"
  "${CMAKE_CURRENT_LIST_DIR}/test/ut/main.cpp"
)
set(UT_MOD_DEPS STest)
register_fprime_ut()
```

After that you might want to try removing the existing cache:
```sh
cd Ref
rm -r build-fprime-automatic-native-ut
fprime-util generate --ut
cd MathSender
fprime-util impl --ut
...
```

capsulecorplab, 2022-02-01:
@astroesteban good catch! Added the missing standard build section, 
```
# Register the standard build
set(SOURCE_FILES
  "${CMAKE_CURRENT_LIST_DIR}/MathSender.cpp"
  "${CMAKE_CURRENT_LIST_DIR}/MathSender.fpp"
)
register_fprime_module()
```
as per https://github.com/capsulecorplab/fprime/commit/38079be72ad2c7befa1113e1be07fe1b9f8a0e4f, then removed `build-fprime-automatic-native-ut` and reran `fprime-util generate --ut`, but now I'm encountering a slightly different CMake error
```
CMake Error at /home/captainpike/repo/fork/fprime/cmake/target/build.cmake:54 (add_executable):
  Cannot find source file:

    /home/captainpike/repo/fork/fprime/Ref/MathSender/test/ut/Tester.cpp

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx
Call Stack (most recent call first):
  /home/captainpike/repo/fork/fprime/cmake/target/ut.cmake:61 (setup_build_module)
  /home/captainpike/repo/fork/fprime/cmake/target/target.cmake:127 (add_module_target)
  /home/captainpike/repo/fork/fprime/cmake/target/target.cmake:143 (setup_module_target)
  /home/captainpike/repo/fork/fprime/cmake/module.cmake:86 (setup_all_module_targets)
  /home/captainpike/repo/fork/fprime/cmake/API.cmake:427 (generate_ut)
  MathSender/CMakeLists.txt:15 (register_fprime_ut)

CMake Error at /home/captainpike/repo/fork/fprime/cmake/target/build.cmake:54 (add_executable):
  No SOURCES given to target: Ref_MathSender_ut_exe
Call Stack (most recent call first):
  /home/captainpike/repo/fork/fprime/cmake/target/ut.cmake:61 (setup_build_module)
  /home/captainpike/repo/fork/fprime/cmake/target/target.cmake:127 (add_module_target)
  /home/captainpike/repo/fork/fprime/cmake/target/target.cmake:143 (setup_module_target)
  /home/captainpike/repo/fork/fprime/cmake/module.cmake:86 (setup_all_module_targets)
  /home/captainpike/repo/fork/fprime/cmake/API.cmake:427 (generate_ut)
  MathSender/CMakeLists.txt:15 (register_fprime_ut)

CMake Generate step failed.  Build files cannot be regenerated correctly.
[ERROR] CMake erred with return code 1. Partial build cache remains. Run purge to clean-up.
```

astroesteban, 2022-02-01:
It looks like you might be missing the placeholder tester files:
```sh
touch test/ut/Tester.cpp
touch test/ut/main.cpp
```

They can just be empty. You'll replace them once `fprime-util impl --ut` works and generates the actual files.
It's just so that CMake doesn't complain.

capsulecorplab, 2022-02-01:
@astroesteban Ah, that was it! Thank you!!

closed 2022-02-01 by capsulecorplab

astroesteban, 2022-02-01:
Happy to help @capsulecorplab! 🚀

capsulecorplab, 2022-02-01:
Hm, now I'm getting an error with `fprime-util build --ut`,
```
In file included from /home/captainpike/repo/fork/fprime/Ref/MathSender/test/ut/Tester.cpp:13:
/home/captainpike/repo/fork/fprime/Ref/MathSender/test/ut/Tester.hpp:17:10: fatal error: Ref/MathSender/MathSenderComponentImpl.hpp: No such file or directory
   17 | #include "Ref/MathSender/MathSenderComponentImpl.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [Ref/MathSender/CMakeFiles/Ref_MathSender_ut_exe.dir/build.make:93: Ref/MathSender/CMakeFiles/Ref_MathSender_ut_exe.dir/test/ut/Tester.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:33363: Ref/MathSender/CMakeFiles/Ref_MathSender_ut_exe.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:33370: Ref/MathSender/CMakeFiles/Ref_MathSender_ut_exe.dir/rule] Error 2
make: *** [Makefile:11940: Ref_MathSender_ut_exe] Error 2
[ERROR] CMake erred with return code 2
```

capsulecorplab, 2022-02-03:
Not sure if the aforementioned build error is related, so opened a new issue in https://github.com/nasa/fprime/issues/1258