nasa/fprime #1244: fprime-util new cmakelist addition error

legounicycler, 2022-01-29:
| | |
|:---|:---|
|**_F´ Version_**|2.0.1 |
|**_Affected Component_**|  fprime-util new |
---
## Problem Description

When using the `fprime-util new` command to create a new component, at the very end of the process it will attempt to add a line to your deployment's `CMakeLists.txt` for the new component. However, when you run this command OUTSIDE the deployment folder and use the `-d` flag, the command will successfully create the files associated with the component, but it will error on the step where it adds the line to the deployment `CMakeLists.txt`

Repository structure in question:

```
gt2-software (project root)
    |
    |-- fprime (submodule)
    |
    |-- Primary (cookiecutter deployment)
    |
    |-- Common (The folder containing your fprime component)
```

## How to Reproduce

1. In a repository in which fprime is included as a submodule, create a deployment at the same level as the fprime directory. See the repository structure above as an example. (In my case, to generate the deployment labeled "Primary," I used Sterling Peet's fprime deployment cookiecutter located here: https://github.com/SterlingPeet/cookiecutter-fprime-deployment ).
2. Run `fprime-util generate` in the deployment folder. (In my case, "Primary").
3. Create a new component outside the deployment folder using `fprime-util new -d <path-to-your-deployment>` (In my case, I ran `fprime-util new -d ../Primary` from the `Common` directory.)
4. Walk through the prompts and hit enter for all default options.
5. Component folder and files will successfully generate, but you will receive an error like this where you would normally get a prompt asking if you'd like to add your component to the CMakeList: `ValueError: '/home/anthony/fprime-devel/gt2-software/Common/GPS' does not start with '../Primary'`

## Expected Behavior

`fprime-util new` will automatically detect the CMakeLists.txt file within your deployment folder (in the below example, the deployment is called  "Primary")
You will then be prompted to include the component in the topology and generate implementations as is shown below.

```[INFO] Found CMakeLists.txt at '/home/anthony/fprime-devel/gt2-software/Primary/CMakeLists.txt'
[INFO] Topology inclusion 'add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/Top/")' found on line 80.
Add component GPS to /home/anthony/fprime-devel/gt2-software/Primary/CMakeLists.txt  before topology inclusion?yes
Generate implementations and merge into /home/anthony/fprime-devel/gt2-software/Primary/GPS/GPSComponentImpl.hpp and /home/anthony/fprime-devel/gt2-software/Primary/GPS/GPSComponentImpl.cpp?yes```

ThibFrgsGmz, 2022-01-29:
This must be followed in fprime-util issue tracker, right? 
Now that v3 with FPP is out, I guess no attention will be paid to its resolution?

LeStarch, 2022-02-08:
F´ new is still something we support.  It does need an update for FPP and the bug described here also needs to be fixed.

LeStarch, 2022-09-26:
Linking this issue to the `fprim-util new` issue and closing.

closed 2022-09-26 by LeStarch