nasa/fprime #1248: fprime-util check --coverage fails

kevin-f-ortega, 2022-01-31:
| | |
|:---|:---|
|**_F´ Version_**| 7e20febd34019c730da1358567e7a512592de4d8|
|**_Affected Component_**| Code Coverage  |
---
## Problem Description

When running code coverage, it fails with the following

```
100% tests passed, 0 tests failed out of 1

Total Test time (real) =   0.31 sec
[100%] Built target Libraries_Util_check
/home/.../Libraries_Util.dir/AtomicUtil.gcno:cannot open notes file
/home/.../Libraries_Util.dir/AtomicUtil.gcda:cannot open data file, assuming not executed
/home/.../Libraries_Util.dir/Util.gcno:cannot open notes file
/home/.../Libraries_Util.dir/Util.gcda:cannot open data file, assuming not executed
/home/.../Libraries_Util.dir/BitUtil.gcno:cannot open notes file
/home/.../Libraries_Util.dir/BitUtil.gcda:cannot open data file, assuming not executed
Error copying file "*.gcov" to "/home/.../Libraries/Util/coverage".
make[3]: *** [Libraries/Util/CMakeFiles/Libraries_Util_coverage.dir/build.make:72: Libraries/Util/CMakeFiles/Libraries_Util_coverage] Error 1
make[2]: *** [CMakeFiles/Makefile2:32568: Libraries/Util/CMakeFiles/Libraries_Util_coverage.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:32575: Libraries/Util/CMakeFiles/Libraries_Util_coverage.dir/rule] Error 2
make: *** [Makefile:12204: Libraries_Util_coverage] Error 2
[ERROR] CMake erred with return code 2
```

## How to Reproduce

1. fprime-util check --coverage

## Expected Behavior

I expected the coverage files to be generated and in the coverage directory.

LeStarch, 2022-02-02:
This is a problem with the `gcov` executable.  It can only run on a single directory, so modules containing multiple directories break.

e.g. Os is one module across multiple folders:

```
Os/*.cpp
Os/Posix/*.cpp
...
```

This results in an output of .o and .gcno files in a mirrored structure:

```
Os/*.o
Os/*.gcno
Os/Posix/*.o
Os/Posix/*.gcno
```

However, when F´ runs gcov, it cannot handle the multiple locations for the .gcno outputs.

LeStarch, 2022-02-02:
There are two fixes:

1. Make cmake detect and run gcov on every possible location one at a time
2. Switch to a wrapper tool (e.g. gcovr) that performs this magic on our behalf

closed 2022-04-28 by LeStarch