nasa/fprime #1251: F´ Terminology Documentation

LeStarch, 2022-02-01:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Write a guide for translating "software engineering concepts"  to "F´nomenclature".  

e.g.

Queue -> Os::Queue or Async Port call
Map -> Table + for loop

LeStarch, 2022-03-22:
Also include CMake terms and usage.

matt392code, 2024-12-27:
# F Prime Translation Guide: Software Engineering to F Prime Nomenclature

## Data Structures & Containers

| Software Concept | F Prime Equivalent | Notes |
|-----------------|-------------------|--------|
| Queue | `Os::Queue` | Thread-safe queue implementation for RTOS |
| Array | `Fw::Array<T>` | Fixed-size array with bounds checking |
| String | `Fw::String` | Safe string implementation with size limits |
| Buffer | `Fw::Buffer` | Memory buffer with size tracking |
| Vector | `Fw::Array<T>` | F Prime uses fixed arrays instead of dynamic vectors |
| Map/Dictionary | Table + for loop | Implement lookup tables with arrays and linear search |
| Set | Bit array/mask | Use bit fields for efficient set operations |

## Communication & Synchronization

| Software Concept | F Prime Equivalent | Notes |
|-----------------|-------------------|--------|
| Function Call | Synchronous Port | Direct component-to-component calls |
| Callback | Port Registration | Components register ports for callbacks |
| Event Loop | Active Component | Components with their own execution thread |
| Message Queue | Async Port | Asynchronous component communication |
| Publish/Subscribe | Event Port | Multi-cast port connections |
| Mutex | `Os::Mutex` | Thread synchronization primitive |
| Semaphore | `Os::Semaphore` | Resource management between threads |
| Thread | `Os::Task` | RTOS task abstraction |
| Timer | Rate Group | Time-based execution scheduling |

## Memory Management

| Software Concept | F Prime Equivalent | Notes |
|-----------------|-------------------|--------|
| Heap Allocation | `Fw::MemAllocator` | Managed heap allocation |
| Stack Allocation | Local variables | Standard stack allocation |
| Memory Pool | `Svc::BufferManager` | Fixed-size buffer management |
| Smart Pointer | Object ownership | Components own their objects |
| Reference Counting | Port references | Managed by component framework |
| Memory Mapping | `Drv::BlockDriver` | Hardware memory management |

## System Architecture

| Software Concept | F Prime Equivalent | Notes |
|-----------------|-------------------|--------|
| Module | Component | Basic unit of functionality |
| Interface | Port | Component communication interface |
| Service | Service Component | Components providing system services |
| Driver | Driver Component | Hardware abstraction components |
| Configuration | Parameters | Component configuration management |
| Plugin | Component Library | Reusable component implementations |
| State Machine | Component States | Internal component state management |

## Error Handling

| Software Concept | F Prime Equivalent | Notes |
|-----------------|-------------------|--------|
| Exception | Assert + Event | Assertion and event logging |
| Error Code | Status Type | Enumerated status returns |
| Try/Catch | Command Handler | Command execution with status return |
| Logging | Events | System event logging framework |
| Debug Print | `Fw::Logger` | Debug output facility |

## Testing

| Software Concept | F Prime Equivalent | Notes |
|-----------------|-------------------|--------|
| Unit Test | GTest Component | Google Test integration |
| Mock Object | Test Component | Component test implementations |
| Test Fixture | Test Harness | Component test setup |
| Integration Test | Integration Test | System-level test scenarios |
| Test Coverage | gcov + STest | Code coverage analysis |
[... 82 lines cut ...]
1. **Avoid Direct Memory Management**
   - Use F Prime buffer management
   - Don't use raw new/delete

2. **Port Connection Issues**
   - Check port numbers match
   - Verify sync vs async usage
   - Ensure proper initialization

3. **Resource Limitations**
   - Consider stack size limits
   - Monitor queue depths
   - Watch for buffer overflow

4. **Thread Safety**
   - Use proper synchronization
   - Avoid shared state
   - Consider component activation

Remember that F Prime is designed for embedded systems, so always consider:
- Resource constraints
- Real-time requirements
- Safety considerations
- Deterministic behavior
[fprime-translation-guide.md](https://github.com/user-attachments/files/18256356/fprime-translation-guide.md)

matt392code, 2024-12-27:
# F Prime Translation Guide: Software Engineering to F Prime Nomenclature

## Data Structures & Containers

| Software Concept | F Prime Equivalent | Notes |
|-----------------|-------------------|--------|
| Queue | `Os::Queue` | Thread-safe queue implementation for RTOS |
| Array | `Fw::Array<T>` | Fixed-size array with bounds checking |
| String | `Fw::String` | Safe string implementation with size limits |
| Buffer | `Fw::Buffer` | Memory buffer with size tracking |
| Vector | `Fw::Array<T>` | F Prime uses fixed arrays instead of dynamic vectors |
| Map/Dictionary | Table + for loop | Implement lookup tables with arrays and linear search |
| Set | Bit array/mask | Use bit fields for efficient set operations |

## Communication & Synchronization

| Software Concept | F Prime Equivalent | Notes |
|-----------------|-------------------|--------|
| Function Call | Synchronous Port | Direct component-to-component calls |
| Callback | Port Registration | Components register ports for callbacks |
| Event Loop | Active Component | Components with their own execution thread |
| Message Queue | Async Port | Asynchronous component communication |
| Publish/Subscribe | Event Port | Multi-cast port connections |
| Mutex | `Os::Mutex` | Thread synchronization primitive |
| Semaphore | `Os::Semaphore` | Resource management between threads |
| Thread | `Os::Task` | RTOS task abstraction |
| Timer | Rate Group | Time-based execution scheduling |

## Memory Management

| Software Concept | F Prime Equivalent | Notes |
|-----------------|-------------------|--------|
| Heap Allocation | `Fw::MemAllocator` | Managed heap allocation |
| Stack Allocation | Local variables | Standard stack allocation |
| Memory Pool | `Svc::BufferManager` | Fixed-size buffer management |
| Smart Pointer | Object ownership | Components own their objects |
| Reference Counting | Port references | Managed by component framework |
| Memory Mapping | `Drv::BlockDriver` | Hardware memory management |

## System Architecture

| Software Concept | F Prime Equivalent | Notes |
|-----------------|-------------------|--------|
| Module | Component | Basic unit of functionality |
| Interface | Port | Component communication interface |
| Service | Service Component | Components providing system services |
| Driver | Driver Component | Hardware abstraction components |
| Configuration | Parameters | Component configuration management |
| Plugin | Component Library | Reusable component implementations |
| State Machine | Component States | Internal component state management |

## Error Handling

| Software Concept | F Prime Equivalent | Notes |
|-----------------|-------------------|--------|
| Exception | Assert + Event | Assertion and event logging |
| Error Code | Status Type | Enumerated status returns |
| Try/Catch | Command Handler | Command execution with status return |
| Logging | Events | System event logging framework |
| Debug Print | `Fw::Logger` | Debug output facility |

## Testing

| Software Concept | F Prime Equivalent | Notes |
|-----------------|-------------------|--------|
| Unit Test | GTest Component | Google Test integration |
| Mock Object | Test Component | Component test implementations |
| Test Fixture | Test Harness | Component test setup |
| Integration Test | Integration Test | System-level test scenarios |
| Test Coverage | gcov + STest | Code coverage analysis |
[... 81 lines cut ...]

1. **Avoid Direct Memory Management**
   - Use F Prime buffer management
   - Don't use raw new/delete

2. **Port Connection Issues**
   - Check port numbers match
   - Verify sync vs async usage
   - Ensure proper initialization

3. **Resource Limitations**
   - Consider stack size limits
   - Monitor queue depths
   - Watch for buffer overflow

4. **Thread Safety**
   - Use proper synchronization
   - Avoid shared state
   - Consider component activation

Remember that F Prime is designed for embedded systems, so always consider:
- Resource constraints
- Real-time requirements
- Safety considerations
- Deterministic behavior

LeStarch, 2025-02-14:
@matt392code these tables you posted are exactly what we need for this issue.  Do you want to pull those tables into a PR?  

Create:
```
docs/user-guide/framework/fprime-translations.md
```

Add all of the tables you put at the top (everything up to "Example Translations").  

I will need to go through and edit some of the data, but want to let you get credit for the initial work.

PR #3260 (2025-02-27): Add F Prime terminology translation guide
"Resolves #1251 - Creates documentation to translate software engineering concepts to F´ nomenclature"
LeStarch on Feb 13, 2025
@matt392code these tables you posted are exactly what we need for this issue. Do you want to pull those tables into a PR? Create:
docs/user-guide/framework/fprime-translations.md
Add all of the tables you put at the top (everything up to "Example Translations"). I will need to go through and edit some of the data, but want to let you get credit for the initial work.
| | |
|:---|:---|
|**_Related Issue(s)_**| #1251 |
|**_Has Un [... cut]
closed 2025-02-27 by LeStarch