nasa/fprime #1255: fpp-check Resulting in Error Message

Emil808, 2022-02-02:
| | |
|:---|:---|
|**_F´ Version_**| 3.0.0 |
|**_Affected Component_**| Math Component Tutorial |
---
## Problem Description
Following the tutorial, at section 6.2, Checking for unconnected ports. Command:
```
fprime-util fpp-check -u unconnected.txt
```
is resulting in an error message:
```
[WARNING] Failed to find settings file: <path_to>/fprime/Ref/settings.ini
[INFO] Updating fpp locations file and build cache. This may take some time.
[ERROR] No fpp dependency memo found. Perhaps directory is outside build?
```
Did attempt with purging and generating, error message still shows. 

## How to Reproduce

1. Follow Math Component Tutorial
2. Get to section 6.2 Updating the Topology
3. do command: fprime-util fpp-check -u unconnected.txt

## Expected Behavior
No error message.

astroesteban, 2022-02-09:
I came across the same error. I just ran `fprime-util fpp-check -u unconnected.txt` in the `Ref/Top` directory and that worked.

LeStarch, 2022-02-09:
That is a good catch!  Yes, this is the correct answer.  I will close this issue.

closed 2022-02-09 by LeStarch

Emil808, 2022-02-11:
Okay, I tried it out on Ref/Top, but now im getting this: 
```
[INFO] Updating fpp locations file and build cache. This may take some time.
Traceback (most recent call last):
  File "/usr/local/bin/fprime-util", line 8, in <module>
    sys.exit(main())
  File "/home/emil/.local/lib/python3.8/site-packages/fprime/util/__main__.py", line 14, in main
    return fprime.util.build_helper.utility_entry(args=sys.argv[1:])
  File "/home/emil/.local/lib/python3.8/site-packages/fprime/util/build_helper.py", line 164, in utility_entry
    runners[parsed.command](build, parsed, cmake_args, make_args)
  File "/home/emil/.local/lib/python3.8/site-packages/fprime/fpp/cli.py", line 71, in run_fpp_check
    run_fpp_util(
  File "/home/emil/.local/lib/python3.8/site-packages/fprime/fpp/common.py", line 109, in run_fpp_util
    return subprocess.run(app_args, capture_output=False)
  File "/usr/lib/python3.8/subprocess.py", line 493, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'fpp-check'
```
I thought it was a

LeStarch, 2022-02-11:
This is likely a different error. `fpp-check` was probably installed using the CMake automatic install.....but that doesn't place it on the path where other tools can find it.

In our next release, we will no longer do the automatic install but rather provide an easy way to install FPP in general.  At that point this issue should go away.

danjwait, 2022-02-15:
Sorry to comment on a closed issue; @LeStarch should I open a new issue per the second error @Emil808 reported? I just ran into the same thing: 
```
~/02_Projects/fprime/Ref/Top$ fprime-util fpp-check -u unconnected.txt
[WARNING] Failed to find settings file: /home/djwait/02_Projects/fprime/Ref/settings.ini
[INFO] Updating fpp locations file and build cache. This may take some time.
Traceback (most recent call last):
  File "/home/djwait/.local/bin/fprime-util", line 8, in <module>
    sys.exit(main())
  File "/home/djwait/.local/lib/python3.8/site-packages/fprime/util/__main__.py", line 14, in main
    return fprime.util.build_helper.utility_entry(args=sys.argv[1:])
  File "/home/djwait/.local/lib/python3.8/site-packages/fprime/util/build_helper.py", line 164, in utility_entry
    runners[parsed.command](build, parsed, cmake_args, make_args)
  File "/home/djwait/.local/lib/python3.8/site-packages/fprime/fpp/cli.py", line 71, in run_fpp_check
    run_fpp_util(
  File "/home/djwait/.local/lib/python3.8/site-packages/fprime/fpp/common.py", line 109, in run_fpp_util
    return subprocess.run(app_args, capture_output=False)
  File "/usr/lib/python3.8/subprocess.py", line 493, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'fpp-check'
```
If nothing else then that would be a place to put any work around?

Thanks/sorry!

commit 595cf368 (2022-02-09): astroesteban: fix math component misleading instructions (#1273)

* astroesteban: fix math component tutorial typos

* Minor Notation Fixes
#1255
commit b65e012a (2022-06-29): astroesteban: fix math component misleading instructions (#1273)

* astroesteban: fix math component tutorial typos

* Minor Notation Fixes
#1255