nasa/fprime #1258: Fatal error with`fprime-util build --ut`

capsulecorplab, 2022-02-03:
| | |
|:---|:---|
|**_F´ Version_**| v3.0.0|
|**_Affected Component_**| MathSender |
---
## Problem Description

Fatal error encountered when running `fprime-util build --ut`, following [4.5.1. Set Up the Unit Test Environment](https://nasa.github.io/fprime/Tutorials/MathComponent/Tutorial.html#The-MathSender-Component_Write-and-Run-Unit-Tests_Set-Up-the-Unit-Test-Environment) in [Math Components Tutorial](https://nasa.github.io/fprime/Tutorials/MathComponent/Tutorial.html), as of https://github.com/capsulecorplab/fprime/commit/ce8836c122715a4e79b989e18320e058b40702d3

```
In file included from /home/captainpike/repo/fork/fprime/Ref/MathSender/test/ut/Tester.cpp:13:
/home/captainpike/repo/fork/fprime/Ref/MathSender/test/ut/Tester.hpp:17:10: fatal error: Ref/MathSender/MathSenderComponentImpl.hpp: No such file or directory
   17 | #include "Ref/MathSender/MathSenderComponentImpl.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [Ref/MathSender/CMakeFiles/Ref_MathSender_ut_exe.dir/build.make:93: Ref/MathSender/CMakeFiles/Ref_MathSender_ut_exe.dir/test/ut/Tester.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:33363: Ref/MathSender/CMakeFiles/Ref_MathSender_ut_exe.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:33370: Ref/MathSender/CMakeFiles/Ref_MathSender_ut_exe.dir/rule] Error 2
make: *** [Makefile:11940: Ref_MathSender_ut_exe] Error 2
[ERROR] CMake erred with return code 2
```

## How to Reproduce

### Setup/configuration
On Ubuntu 20.04 with python3,
1. `git clone git@github.com:capsulecorplab/fprime.git`
2. `cd fprime`
3. `git checkout math-tutorial`
4. `python3 -m venv fprime-venv`
5. `. ./fprime-venv/bin/activate`
6. `pip install --upgrade fprime-tools fprime-gds`

### Build steps
In fprime repo with virtual environment activated, 
1. `cd Ref`
2. `fprime-util generate --ut`
2. `cd MathSender`
3. `fprime-util impl --ut`
4. `fprime-util build --ut`

## Expected Behavior

Unit tests should build without errors.

LeStarch, 2022-02-03:
https://github.com/capsulecorplab/fprime/blob/ce8836c122715a4e79b989e18320e058b40702d3/Ref/MathSender/MathSender.hpp

Note: the file name is `MathSender.hpp`  and the UT implementation is assuming `MathSenderComponentImpl.hpp`.   We should ensure the following:

1. Tutorial acknowledges this and ensures users check consistency
2. `fprime-util new`, `fprime-util impl`, and `fprime-util impl --ut` generate consistent files

Thanks for the report!

astroesteban, 2022-02-04:
See https://github.com/nasa/fprime/pull/1243#issuecomment-1030341052

capsulecorplab, 2022-02-08:
@astroesteban Thanks, again! `fprime-util build --ut` seems to be running the unit test build without errors now as of https://github.com/capsulecorplab/fprime/commit/5a815b4d6766edf7db385354688149462790c87d. Will proceed to [4.5.2. Write and Run One Test](https://nasa.github.io/fprime/Tutorials/MathComponent/Tutorial.html#The-MathSender-Component_Write-and-Run-Unit-Tests_Write-and-Run-One-Test) and report back if I run into any more issues

closed 2022-02-08 by capsulecorplab