nasa/fprime #1266: Unexpected version from fpp.cmake

juliannerice, 2022-02-08:
| | |
|:---|:---|
|**3.0.0**| |
|**fpp-tools**|   |
---
## Problem Description
"Not expected version" error message after running _fprime-util generate_ (see picture below)

## How to Reproduce

1. Follow Latest Documentation for F Prime Installation.
2. Under _Ref_ enter_fprime-util generate_ command

## Expected Behavior
[image]

LeStarch, 2022-02-08:
Note this is an error with the automated install, needs urgent fix.

LeStarch, 2022-02-08:
@juliannerice does this affect the latest `devel` or `v3.0.0`?   Do you have any modifications or is this fprime without modifications?

juliannerice, 2022-02-08:
@LeStarch This was upon initial installation without any previous modifications. I was following the latest installation instructions step-by-step when I stumbled upon this. I imagine it's latest devel because I tried cloning from the main github address (https://github.com/nasa/fprime).

astroesteban, 2022-02-08:
Note that this is a relevant conversation:
https://github.com/nasa/fprime/discussions/1235

My temporary fix is to change the commit hash in fpp.cmake to the hash of my FPP version. This issue arises in v3.0.0 and devel for me.

LeStarch, 2022-02-08:
I am struggling to reproduce this issue.  I tried in `nasa/devel` on both macOS and Ubuntu.  Could you give me your CMake versions number?

astroesteban, 2022-02-08:
I am running an Ubuntu 20.04 Docker image where I install FPP as follows:
```docker
# install fpp
WORKDIR /workspace/fpp
RUN git clone https://github.com/fprime-community/fpp.git . \
    && git checkout -b release/1.0.0 v1.0.0 \
    && cd compiler \
    && ./install /usr/local/bin/fpp
ENV FPP_INSTALL_DIR=/usr/local/bin/fpp
ENV PATH=${PATH}:${FPP_INSTALL_DIR}
```

This installs `cmake version 3.16.3`

Then on a fresh clone of fprime to the `devel` branch I run `fprime-util generate` in the `Ref` folder and get the error:
```sh
-- [fpp-tools] Searching for fpp-tools
-- [fpp-tools] /usr/local/bin/fpp/fpp-depend version v1.0.0 not expected version v1.0.0-24-g52b996cb
-- [fpp-tools] Searching for sbt
-- [fpp-tools] sbt found at: /usr/bin/sbt
CMake Error at /workspaces/fprime/cmake/fpp-download/fpp.cmake:81 (message):
  Build step for fpp failed: 2

  Cloning into 'src'...

  Note: switching to 'v1.0.0-24-g52b996cb'.

  

  You are in 'detached HEAD' state.  You can look around, make experimental

  changes and commit them, and you can discard any commits you make in this

  state without impacting any branches by switching back to a branch.

  

  If you want to create a new branch to retain commits you create, you may

  do so (now or later) by using -c with the switch command.  Example:

  

    git switch -c <new-branch-name>

  

  Or undo this operation with:

  

    git switch -

  

  Turn off this advice by setting config variable advice.detachedHead to
  false

  

  HEAD is now at 52b996cb Revise User's Guide

  [info] [launcher] getting org.scala-sbt sbt 1.6.1 (this may take some
  time)...

  make[2]: *** [CMakeFiles/fpp-tools.dir/build.make:75:
  fpp-tools-prefix/src/fpp-tools-stamp/fpp-tools-install] Error 2
-- [fpp-tools] Installing new tools to /workspaces/fprime/Ref/build-fprime-automatic-native/fpp-tools-install, this will take a few minutes
-- Configuring incomplete, errors occurred!

See also "/workspaces/fprime/Ref/build-fprime-automatic-native/CMakeFiles/CMakeOutput.log".
  make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/fpp-tools.dir/all] Error
  2

  make: *** [Makefile:84: all] Error 2

Call Stack (most recent call first):
  /workspaces/fprime/cmake/required.cmake:20 (include)
  /workspaces/fprime/cmake/FPrime.cmake:26 (include)
  CMakeLists.txt:32 (include)

[ERROR] CMake erred with return code 1. Partial build cache remains. Run purge to clean-up.
```

I have my full Dockerfile defined here: https://github.com/astroesteban/fprime-devcontainer/blob/main/.devcontainer/Dockerfile

What's your environment like @juliannerice?

LeStarch, 2022-02-08:
@astroesteban These are two separate errors.   Yours is the failure to install FPP tools within a Docker environment.  The error here is a failure to match the version of the tools that did install correctly.

astroesteban, 2022-02-08:
Ah my apologies. I will delete my above comment.

LeStarch, 2022-02-08:
Didn't need to delete the comment, just clarifying that I am tracking separate issues.

LeStarch, 2022-02-08:
@juliannerice  (or anyone else who sees this) one thing that might help me diagnose would be to do this:

```
cd build-fprime-automatic-native/fpp-tools-install
./fpp-depend --help
```
And post the version it prints as the second line. It is expected to look like:
```
fpp-depend v1.0.0-24-g52b996cb
```
I need to know if the following version appears in some environments. Notice there is no `-b` on the end.

```
fpp-depend v1.0.0-24-g52b996c
```

If the following output is in fact missing the missing `b`, then we know the bad version stamp was introduced into the tools build.  Then we should check the inputs to the build via:

```
git --version
cd cd build-fprime-automatic-native/fpp-tools/src
git describe --tag --always
```

For the curious, I am trying to determine the cause from several possibilities:

1. A cmake version is not handling regex matches in a standard way
2. The tools are not reporting the correct version
3. The version handed to the tools was incorrect

juliannerice, 2022-02-09:
@LeStarch the CMake version number I'm using is 3.17.5

`cmake3 version 3.17.5`

After running the first two commands suggested

`cd build-fprime-automatic-native/fpp-tools-install`

`./fpp-depend --help`

The following is the output:
`fpp-depend v1.0.0-24-g52b996c`

The output after _git --version_ is:
`git version 1.8.3.1`

The output after _git describe --tag --always_ under _build-fprime-automatic-native/fpp-tools/src_ was:
`v1.0.0-24-g52b996c`

LeStarch, 2022-03-22:
This issue seems to be a missmatch between what git outputs, and what fpp-depend outputs.   We have taken the following corrective actions for the next delivery:

1. Adding in a flag to skip this version checking
2. Publishing fixed versions for work with master and stable tags

Not sure what else we can do to fix this.

LeStarch, 2022-04-11:
Closing this issue as the above fixes have been submitted.

closed 2022-04-11 by LeStarch