nasa/fprime #1289: Apply uniform convention to serialize() methods

JackNWhite, 2022-02-22:
| | |
|:---|:---|
|**_F´ Version_**|2.1.0 |
|**_Affected Component_**| At least `TlmChan` and `Serializable`  |
---
## Feature Description
Ensure that methods called in the form `foo.serialize(bar)` always (uniformly across F') either:
- Serialise `foo` and place it in `bar`
- Serialise `bar` and place it in `foo`

## Rationale

I am tracing the path of a telemetry channel from source component to ground, using Ref for simplicity.

In `MathReceiverComponentBase::tlmWrite_MR_FACTOR1()`, we see this line:
```cpp
Fw::SerializeStatus _stat = _tlmBuff.serialize(arg);
```
... which appears to me like `arg` is being serialised and stored in a byte array in `_tlmBuff`. That makes sense.

However, in `TlmChanImpl::Run_handler()` we see this:
```cpp
Fw::SerializeStatus stat = this->m_tlmPacket.serialize(this->m_comBuffer);
...
this->PktSend_out(0,this->m_comBuffer,0);
```

In the first case, `arg` is serialised and stored in `_tlmBuffer`; in the second, `this->m_tlmPacket` is serialised and stored in `this->m_comBuffer`.

This is chaotic and contradictory.

timcanham, 2022-02-22:
It works, but it also works if you do this:

```c++
                Fw::SerializeStatus stat = this->m_comBuffer.serialize(this->m_tlmPacket);
                FW_ASSERT(Fw::FW_SERIALIZE_OK == stat,static_cast<NATIVE_INT_TYPE>(stat));
                p_entry->updated = false;
                this->PktSend_out(0,this->m_comBuffer,0);
```

It would be clearer to do it this way. There are some upcoming changes we can roll this is.

matt392code, 2025-01-02:
Suggested serialization documentation:
[serialize-standard.md](https://github.com/user-attachments/files/18294171/serialize-standard.md)
The document covers:
 - The core convention of destination.serialize(source)
 - Detailed implementation requirements
 - Error handling patterns
- Buffer management considerations
- Testing requirements
- Migration guidelines