nasa/fprime #1290: Intermittent UT failure in Drv::TcpServer

kevin-f-ortega, 2022-02-23:
| | |
|:---|:---|
|**_F´ Version_**| v3.0.0 |
|**_Affected Component_**|  Drv::TcpServer |
---
## Problem Description

Intermittent unit test failure in Drv::TcpServer when running `fprime-util  check native --all --jobs 24`

## How to Reproduce

1. cd <deployment-directory>
2. fprime-util check native --all --jobs 24

## Expected Behavior

I expect Drv::TcpServer's UTs to always pass.

kevin-f-ortega, 2022-02-24:
This happened with `Drv::TcpClient` too.

kevin-f-ortega, 2022-02-24:
Drv_p_Tcp_test too

LeStarch, 2022-02-28:
Does this problem persist when running:

```
fprime-util check native --all --jobs 1
```

LeStarch, 2022-03-01:
If errors persist, please post them here.

LeStarch, 2022-11-30:
No reproduction reported.  Closing.  Reopen if specific reproductions available.

closed 2022-11-30 by LeStarch