nasa/fprime #1294: Add Baremetal Compile Step To CI Pipeline

astroesteban, 2022-02-28:
| | |
|:---|:---|
|**_F´ Version_**|v3 |
|**_Affected Component_**| Baremetal & CI |
---
## Feature Description

> Seems like we should add a baremetal compile to CI to prevent this style of error.

_Originally posted by @LeStarch in https://github.com/nasa/fprime/pull/1293#issuecomment-1054470039_

## Rationale

There are some simple bugs like those found in pull requests #1292 and #1293 that could've been easily caught if there was a baremetal compilation step in the CI pipeline.

LeStarch, 2022-03-22:
Nice to have: improve TCP components for "poll" mode such that we can run a polled baremetal system on Linux with an Os to test that the full setup works.

LeStarch, 2024-03-27:
Closed in-favor of #2614 which has an extensive write-up.

closed 2024-03-27 by LeStarch