nasa/fprime #1295: fpp-check, fpp-to-cpp resulting in `error: undefined symbol`

Emil808, 2022-03-01:
| | |
|:---|:---|
|**_F´ Version_**|3.0.0 |
|**_Affected Component_**| fpp-tools  |
---
## Problem Description

Running fpp-check or fpp-to-cpp results in `error undefined symbol`. 

For example: 
```
<>:~/<my_path_to>/fprime/Ref/Top$ fpp-check topology.fpp
fpp-check
<my_path_to>/fprime/Ref/Top/topology.fpp: 24.14
    instance $health
             ^
error: undefined symbol health
```
## How to Reproduce

1. Run `fpp-check` or `fpp-to-cpp` with `/Ref/Top/topology.fpp`

## Expected Behavior

Behavior according to [fpp-check](https://github.com/fprime-community/fpp/wiki/fpp-check) and [fpp-to-cpp](https://github.com/fprime-community/fpp/wiki/fpp-to-cpp).

## 
I'm learning on how to use the fpp-tools to understand how to make a deployment from scratch by testing the fpp-tools on Ref to see how the tools behave. I'm following [this guide](https://fprime-community.github.io/fpp/fpp-users-guide.html#Writing-C-Plus-Plus-Implementations_Implementing-Deployments) on sections _Topology Definitions_ and _Implementing Deployments_.

LeStarch, 2022-03-01:
I think you need to run it as such:

```
fpp-check *.fpp
```

As the topology in Ref is split across several files.  Not you may also need to supply the locations index found after a build at:

<path/to>/Ref/build-fprime-automatic-native/locs.fpp

Emil808, 2022-03-01:
@LeStarch 
I attempted your suggestion and ran this within `/Ref/Top`
```
fpp-check *.fpp <path_to>/Ref/build-fprime-automatic-native/fpp-locs/locs.fpp
```
and now it is giving: 
```
instance blockDrv: Drv.BlockDriver base id 0x0100 \
                     ^
error: undefined symbol Drv
```

LeStarch, 2022-03-01:
This means that you need to supply all the FPPs that go into the model.  You should try running:

```
cd Ref/Top
fprime-util fpp-check
```

As this supplies all the necessary files for you.

Emil808, 2022-03-01:
Neat thanks!
So `fprime-util fpp-check` runs fpp check on the build that was generated which supplies all *.fpp that is related. Then in which context does fpp-check is used?

LeStarch, 2022-03-01:
The fpp tool suite requires  the complete model to process. Thus, you can run them by hand:

```
fpp-check <path/to>/locs.fpp <path/to>/comp1.fpp <path/to>/comp2.fpp <path/to>/topology.fpp
```

However, this quickly becomes tedious as there are a lot of FPP files in a deployment or associated with a topology.  This information (which FPP files compose a topology) is known to the build. `fprime-util fpp-check` calls `fpp-check` but passes in the build's notes so that you don't need to!

`fpp-to-cpp` is automatically run as part of the build, and thus most users would not run this.  You can run it by hand, but again, would need to supply all the necessary files.

If in a CMakeLists.txt file:

```
set(SOURCE_FILES <path/to>/file.fpp)
```

Will run `fpp-to-xml` or `fpp-to-cpp` on the file as appropriate for the build.

Emil808, 2022-03-01:
I see, thanks for the that information!

closed 2022-03-01 by Emil808