nasa/fprime #1296: Assert possible with particular bytes sent to Deframer

abcouwer-jpl, 2022-03-01:
| | |
|:---|:---|
|**_F´ Version_**| Current Master |
|**_Affected Component_**| FprimeProtocol, Deframer  |
---
## Problem Description

https://github.com/nasa/fprime/blob/8a64fb0e4ad44d4a9f7dfb99640e73d5961cf3d2/Svc/FramingProtocol/FprimeProtocol.cpp#L98
https://github.com/nasa/fprime/blob/8a64fb0e4ad44d4a9f7dfb99640e73d5961cf3d2/Svc/Deframer/DeframerComponentImpl.cpp#L125

If the bytes passed to FprimeProtocol in the size slot are such that when adding FP_FRAME_HEADER_SIZE and HASH_DIGEST_LENGTH to size cause a rollover and set needed = 0, then Deframer will do an assert.

## How to Reproduce

This happened when I was trying to modify deframer with larger buffers for ColdArm project's needs. I am not sure if it is due to my own random seed on the randomized unit tests for deframer. It could probably be reproduced by writing a quick script to write the troublesome bytes as input to a deployment's ground interface.

## Expected Behavior

Fprime protocol could check if `size` is > U32_MAX - (FP_FRAME_HEADER_SIZE + HASH_DIGEST_LENGTH) and fix `needed` appropriately, then return DeframingProtocol::DEFRAMING_INVALID_SIZE

LeStarch, 2022-03-01:
Agreed.  We need to make `needed` and `size` resilient against overflow.

bocchino, 2022-03-22:
What should `needed` be set to if the needed value won't fit in 32 bits? It seems to me zero is a reasonable value to use, and maybe Deframer should not fail an assertion in this case.

closed 2022-04-20 by LeStarch