nasa/fprime #1301: Unit Test Error With FW_OBJECT_NAMES Set To Zero

astroesteban, 2022-03-02:
| | |
|:---|:---|
|**_F´ Version_**|v3.0.0 |
|**_Affected Component_**| F Prime CodeGen  |
---
## Problem Description

With `FW_OBJECT_NAMES` set to zero the generated Tester.cpp has an empty constructor:
```cpp
#if FW_OBJECT_NAMES == 1
      BlinkerGTestBase("Tester", MAX_HISTORY_SIZE),
      component("Blinker")
#else
      BlinkerGTestBase(MAX_HISTORY_SIZE),
      component()
#endif
```

but that constructor is not defined in the component's implementation:
```cpp
namespace TivaBlink {

  class Blinker :
    public BlinkerComponentBase
  {

    public:

      // ----------------------------------------------------------------------
      // Construction, initialization, and destruction
      // ----------------------------------------------------------------------

      //! Construct object Blinker
      //!
      Blinker(
          const char *const compName /*!< The component name*/
      );
```

The TesterBase.cpp also has the same issue:
```cpp
#if FW_OBJECT_NAMES == 1
      Fw::PassiveComponentBase(compName)
#else
      Fw::PassiveComponentBase()
#endif
```

This results in a build error when trying to compile the unit tests.

## How to Reproduce

1. I defined a component in FPP:
```
module TivaBlink {
    @ Component for blinking a tiva launchpad's onboard led
    passive component Blinker {
        @ Schedule input port
        sync input port schedIn: Svc.Sched
    }
}
```

2. In my FpConfig.hpp file I have:
```cpp
#define FW_OBJECT_NAMES                     0
```

> I pretty much copied the Arduino's [FpConfig.hpp](https://github.com/fprime-community/fprime-arduino/blob/master/config/FpConfig.hpp)

3. Ran `fprime-util generate --ut`

4. Ran `fprime-util impl --ut`

5. Then ran `fprime-util build --ut`

## Expected Behavior

There's three approaches that I can think of to fix this:

1. Add the empty default constructor to the component implementation and the parent ComponentBase.
2. Have the `Blinker(const char *const compName);` constructor have a default argument: `Blinker(const char *const compName = "");`
3. The simplest approach is to just change `component()` to `component("")` and  `Fw::PassiveComponentBase()` to `Fw::PassiveComponentBase("")` in the test stubs.

LeStarch, 2022-03-08:
A while back we transitioned all constructors to accept names.  FW_OBJECT_NAMES now decides if the name is discarded or stored and reported.

Here, we need to setup the UT templates to pass in a name.  I believe this is your solution 3 above.

astroesteban, 2022-03-24:
@LeStarch Could you point me to the UT templates? For the life of me I can't seem to locate them. Are they in this fprime project?

LeStarch, 2022-03-24:
Tester.cpp: https://github.com/nasa/fprime/blob/devel/Autocoders/Python/src/fprime_ac/generators/templates/test_impl/cpp.tmpl
Tester.hpp: https://github.com/nasa/fprime/blob/devel/Autocoders/Python/src/fprime_ac/generators/templates/test_impl/hpp.tmpl
TesterBase.cpp: https://github.com/nasa/fprime/blob/devel/Autocoders/Python/src/fprime_ac/generators/templates/test/cpp.tmpl
TesterBase.hpp: https://github.com/nasa/fprime/blob/devel/Autocoders/Python/src/fprime_ac/generators/templates/test/hpp.tmpl
TestMain.cpp: https://github.com/nasa/fprime/blob/devel/Autocoders/Python/src/fprime_ac/generators/templates/test/test_main.tmpl
GTestBase.cpp: https://github.com/nasa/fprime/blob/devel/Autocoders/Python/src/fprime_ac/generators/templates/gtest/cpp.tmpl
GTestBase.hpp: https://github.com/nasa/fprime/blob/devel/Autocoders/Python/src/fprime_ac/generators/templates/gtest/hpp.tmpl

astroesteban, 2022-03-24:
Thank you for taking the time to compile this list! I'll look into getting a fix in for this since I need it for my fprime bare metal port.

LeStarch, 2022-03-25:
I would appreciate it!

LeStarch, 2022-04-12:
@astroesteban any progress on this?  We were thinking of including it in an upcoming delivery. Let me know if I can help, or if you'd like to toss this back at us.

astroesteban, 2022-04-12:
Hey @LeStarch I'm gonna need to toss it back to y'all. I have to focus on some unrelated tasks for the time being.

LeStarch, 2022-04-12:
@astroesteban no problem!  Thanks for letting me know!

PR #1389 (2022-04-13): Update/object names 0
| | |
|:---|:---|
|**_Originating Project/Creator_**| |
|**_Affected Component_**|  |
|**_Affected Architectures(s)_**|  |
|**_Related Issue(s)_**|  |
|**_Has Unit Tests (y/n)_**|  |
|**_Builds Without Errors (y/n)_**|  |
|**_Unit Tests Pass (y/n)_**|  |
|**_Documentation Included (y/n)_**|  |

---
## Change Description

Fixes: #1301