nasa/fprime #1303: CMake error encountered following *4.5.4. Exercise: Random Testing* in MathComponent Tutorial

capsulecorplab, 2022-03-03:
| | |
|:---|:---|
|**_F´ Version_**| v3.0.0 |
|**_Affected Component_**| Ref/MathSender |
---
## Problem Description

Following steps in [4.5.4. Exercise: Random Testing](https://nasa.github.io/fprime/Tutorials/MathComponent/Tutorial.html#The-MathSender-Component_Write-and-Run-Unit-Tests_Exercise-Random-Testing), running `fprime-util check` encounters following CMake error,

```
/home/captainpike/repo/fork/fprime/Ref/MathSender/test/ut/main.cpp:2:10: error: #include expects "FILENAME" or <FILENAME>
    2 | #include STest/Random/Random.hpp
      |          ^~~~~
/home/captainpike/repo/fork/fprime/Ref/MathSender/test/ut/main.cpp: In function ‘int main(int, char**)’:
/home/captainpike/repo/fork/fprime/Ref/MathSender/test/ut/main.cpp:16:3: error: ‘STest’ has not been declared
   16 |   STest::Random::seed();
      |   ^~~~~
make[3]: *** [Ref/MathSender/CMakeFiles/Ref_MathSender_ut_exe.dir/build.make:106: Ref/MathSender/CMakeFiles/Ref_MathSender_ut_exe.dir/test/ut/main.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:33363: Ref/MathSender/CMakeFiles/Ref_MathSender_ut_exe.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:33370: Ref/MathSender/CMakeFiles/Ref_MathSender_ut_exe.dir/rule] Error 2
make: *** [Makefile:11940: Ref_MathSender_ut_exe] Error 2
[ERROR] CMake erred with return code 2
```

See https://github.com/capsulecorplab/fprime/commit/21e104b29677f5c402c3fe63852faaf706643279

## How to Reproduce

### Setup/configuration
On Ubuntu 20.04 with python3,
1. `git clone git@github.com:capsulecorplab/fprime.git`
2. `cd fprime`
3. `python3 -m venv fprime-venv`
4. `. ./fprime-venv/bin/activate`
5. `pip install --upgrade fprime-tools fprime-gds`

### Build steps
In fprime repo with virtual environment activated, 
1. `git checkout 21e104b`
2. `cd Ref/MathSender`
3. `fprime-util check`

## Expected Behavior

Unit tests should build without errors.

MichaelStarks, 2022-03-05:
I ran into this problem, the quotes are missing from the include statement in the tutorial doc.

capsulecorplab, 2022-03-06:
Thanks, @MichaelStarks that fixed it

LeStarch, 2022-03-08:
Good catch @MichaelStarks.  Thanks for the PR @capsulecorplab

PR #1308 (2022-03-15): add missing quotes on include statement in math component tutorial section 4.5.4
| | |
|:---|:---|
|**_Originating Project/Creator_**| |
|**_Affected Component_**| Math Component Tutorial |
|**_Affected Architectures(s)_**|  |
|**_Related Issue(s)_**| #1303   |
|**_Has Unit Tests (y/n)_**|  |
|**_Builds Without Errors (y/n)_**|  |
|**_Unit Tests Pass (y/n)_**|  |
|**_Documentation Included (y/n)_**|  |

---
## Change Description

Resolves #1303 

## Rationale

Adds missing quotes on include statement in Math Component Tutorial section [4.5.4 Exercise: Random Testing](https://nasa.github.io/fpr [... cut]
closed 2022-03-15 by LeStarch