nasa/fprime #1329: add hardware interface tutorial

capsulecorplab, 2022-03-18:
| | |
|:---|:---|
|**_F´ Version_**| v3.0.1 |
|**_Affected Component_**|  |
---
## Feature Description

A tutorial for building a component that provides a means to communicate with external hardware, s.a., via I2C

## Rationale

Such a tutorial would be useful for reading external sensors, s.a., temperature sensors, magnetometers, accelerometers, etc.

codeflight1, 2022-03-18:
This already exists in the form of the GPS tutorial: https://nasa.github.io/fprime/Tutorials/GpsTutorial/Tutorial.html

capsulecorplab, 2022-03-19:
Thanks @codeflight1, I see the GPS tutorial is for creating a component that defines ports for supplying buffers to a UART driver. Will this port work for I2C as well?

rawinza555, 2022-05-19:
Not sure if this is too late but I had make F' to use I2C driver and have it worked on raspberry pi 3B+. Note that this is built on F' v.2.0. You can check out my [repo](https://github.com/rawinza555/FPrime_I2C_ADXL345)

LeStarch, 2022-09-26:
This was completed with the [system reference](https://github.com/fprime-community/fprime-system-reference)!  We'll be cleaning up the documentation and layout there over the coming months.  However, future work should is tracked there.

closed 2022-09-26 by LeStarch