nasa/fprime #1339: Even Better/Faster CMake

LeStarch, 2022-03-22:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Building on top of #1260, this is the set of changes that is lower priority after the faster install setup. These (lower priority) items are:

Performance (build):
- [ ] Precompiled headers option investigation
- [ ] Single compilation unit option investigation
- [ ] Precache FPP to XML outputs per-deployment
- [ ] Ensure Ninja functions correctly as underlying make system

astroesteban, 2022-06-28:
Can a user specify if they want to use Ninja or Make in the case where both are installed? Maybe with `fprime-util generate -G Ninja`?

LeStarch, 2022-06-29:
At this time, I don't even know if Ninja works.  If it does work, we'd need to look into how to specify it.  I can see your sugesstion working, or perhaps the more generic:

```
fprime-util generate --cmake-args -G Ninja...
```
So that we can expose the full power of CMake. Thoughts?

astroesteban, 2022-06-29:
So I installed `ninja-build` in Ubuntu 22.04 and then, according to the [CMake docs](https://cmake.org/cmake/help/latest/envvar/CMAKE_GENERATOR.html#envvar:CMAKE_GENERATOR), I set the generator environment variable with `export CMAKE_GENERATOR=Ninja`. It looks like CMake picked it up and then successfully built the Ref deployment with `fprime-util generate` and `fprime-util build`.

Obviously having to export an environment variable isn't the most ideal approach but it works. I like your idea of adding the `--cmake-args` flag. That could just pass any additional arguments straight to CMake.

LeStarch, 2022-09-30:
I have been running in CMake with ninja for some time because that is what the IDE I use does.  No issues.

LeStarch, 2022-09-30:
We should consider also a large restructuring to break apart the large monolith of targets required by modern projects.

sobkulir, 2023-10-14:
I wanted to open a new discussion about Ninja vs Make, but then I found this thread so I'll just post here instead. For our project, Ninja turned out to be ~5x faster than Make. We use Zephyr and its default build system uses Ninja so it probably isn't a coincidence.

However, for me, also the `Ref` deployment builds >2x faster with Ninja vs Make. So I never used Make in the last 6 months. Maybe it's just my setup, is anyone experiencing similar discrepancies between the two generators? If yes, I believe it would make sense to make Ninja the default generator.

My super simple benchmarks:
```shell
# Building Ref first with Ninja and then with Make.

# Ninja
$ time `fprime-util generate -DCMAKE_GENERATOR=Ninja && fprime-util build -j16`
real    0m23.051s
user    1m48.154s
sys     0m26.662s

# Make
$ time `fprime-util generate && fprime-util build -j16`
real    1m5.167s
user    1m30.670s
sys     0m29.443s
```

matt392code, 2025-01-01:
[cmake-improvements.md](https://github.com/user-attachments/files/18284464/cmake-improvements.md)
Proposed structured approach, particularly focusing on the build system optimization.
Key aspects of this proposal:
1. Ninja Integration
- Better dependency tracking
- Faster builds (2-5x improvement)
- Seamless IDE integration
2. Build System Configuration
```bash
# Example usage
fprime-util generate --cmake-args "-GNinja -DFPRIME_USE_PCH=ON"
fprime-util build -j$(nproc)
```
3. Progressive Enhancement
- Features can be enabled independently
- Maintains compatibility
- Clear performance metrics

closed 2025-04-16 by LeStarch