nasa/fprime #1342: strnlen is not standard C++11 and not implemented in fprime

kevin-f-ortega, 2022-03-22:
| | |
|:---|:---|
|**_F´ Version_**| 47afb72e1f2cb79c1ab475458b0ed3faca5d317c|
|**_Affected Component_**|  Any component using strnlen |
---
## Problem Description

strnlen is not part of the C++11 standard.
When compiling with C++11 (without any extensions e.g. gnu++11), the compiler complains that strnlen is not defined.

## How to Reproduce

1.  Use only -std=c++11 without any extensions
2.
3.

## Expected Behavior

strnlen should be defined.

LeStarch, 2022-03-22:
**Note:** we need to implement our own `strnlen` implementation and put it in `StringUtils::string_length()`.  I'd recommend not even creating the `strnlen` function and just putting an implementation directly in `string_length`.

If users want to call `strnlen` from their code, and implementation is up to them. Otherwise, use `StringUtils::string_length()`.

@kevin-f-ortega does this make sense?

kevin-f-ortega, 2022-03-23:
@LeStarch that make sense. Thanks!

PR #1382 (2022-04-12): Update/strnlen replacement
| | |
|:---|:---|
|**_Originating Project/Creator_**| |
|**_Affected Component_**|  |
|**_Affected Architectures(s)_**|  |
|**_Related Issue(s)_**|  |
|**_Has Unit Tests (y/n)_**|  |
|**_Builds Without Errors (y/n)_**|  |
|**_Unit Tests Pass (y/n)_**|  |
|**_Documentation Included (y/n)_**|  |

---
## Change Description

Fixes #1342.
closed 2022-04-12 by LeStarch