nasa/fprime #1366: Seeding STest::Random causes a failure in Deframer unit tests

bocchino, 2022-04-07:
Steps to reproduce:

1. In `Svc/Deframer/test/ut-unified/TestMain.cpp`, add the line `STest::Random::seed();` just before `return RUN_ALL_TESTS();`
2. In `Svc/Deframer`, run `fprime-util build --ut`
3. In `build-fprime-automatic-native-ut/bin/[native os name]`
    1. Run `echo 545994 > seed`. This step captures the random seed that causes the issue to appear.
    2. Run `./Svc_Deframer_with_fprime`. You should see the failure output.

LeStarch, 2022-04-19:
More failing seeds:
386613
774723

bocchino, 2022-04-21:
I think I have a fix for this one. Right now the tests are adding non-uplink data to the received deque on the input side, and then searching through this data on the output side. The tests seem to work better if we just don't add this data on the input side. Then the search code on the output side can go away. I tried it, and these seeds seem to work now.

bocchino, 2022-04-21:
I'm preparing a PR following the unit test review. If I confirm that this fix works, I'll include it in the PR. @arizvi786 @kevin-f-ortega FYI.

PR #1386 (2022-04-13): Fix/deframer ut
| | |
|:---|:---|
|**_Originating Project/Creator_**| |
|**_Affected Component_**|  |
|**_Affected Architectures(s)_**|  |
|**_Related Issue(s)_**|  |
|**_Has Unit Tests (y/n)_**|  |
|**_Builds Without Errors (y/n)_**|  |
|**_Unit Tests Pass (y/n)_**|  |
|**_Documentation Included (y/n)_**|  |

---
## Change Description

Fixes #1366
closed 2022-05-03 by LeStarch