nasa/fprime #1378: FileUplink packet sequence repeat and CRC

timcanham, 2022-04-11:
| | |
|:---|:---|
|**_F´ Version_**| 3.0 |
|**_Affected Component_**| Svc/FileUplink |
---
## Feature Description

Add duplicate packet detection based on sequence number and don't add to CRC computation

## Rationale

If the radio link is marginal, a radio can do a retry and send the same packet. In that case, you get the following warning.

```
FileUplink_PacketOutOfOrder: Received packet 342 after packet 342
```

The repeated packet still gets added to the CRC computation, so even with a successful retry, you get this warning at the end:

```
FileUplink_BadChecksum: Bad checksum value during receipt of file XXXXX: computed 0x6A4F6972, read 0x92A51AAA
```

The update would to be to not update the CRC only in the case of a repeated packet sequence number. This wouldn't be to try to track all cases of dropped and retried packets.

DJKessler, 2024-10-23:
@timcanham -- In the case of a repeated packet, should the repeated packet still get written to the file? In that case, I'll simply protect against updating the checksum by wrapping [THIS LINE](https://github.com/nasa/fprime/blob/2c8d0db8411fe955476925fad8e046e8100653ff/Svc/FileUplink/File.cpp#L56) in a condition. 

If we don't want to write the repeated packet to the file, then I'll wrap [THIS BLOCK OF CODE](https://github.com/nasa/fprime/blob/2c8d0db8411fe955476925fad8e046e8100653ff/Svc/FileUplink/FileUplink.cpp#L131-L138) in a condition that uses the same logic as `checkSequenceIndex()`.

timcanham, 2024-10-23:
@DJKessler I would skip the second write if the first copy was received successfully.

LeStarch, 2024-10-30:
@DJKessler I think this should be reassigned given the schedule change-up.  Do you agree?

DJKessler, 2024-10-30:
@LeStarch -- Schedule change-up? I've actually got this fix implemented. I'm about to submit the pull request.

LeStarch, 2024-11-22:
Closing as complete.

closed 2024-11-22 by LeStarch