nasa/fprime #1385: CI All Things!

LeStarch, 2022-04-12:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

More improvements that can be made to CI!

- [ ] Various FW_CONFIG options
- [ ] Baremetal support

...more will be added

SterlingPeet, 2022-04-14:
reminder to myself that I need to gather up some hardware

ThibFrgsGmz, 2022-04-14:
Maybe it would be interesting to put in the CI the construction of a reference binary from several buildchain.

From what I could see, this is a good practice for software trying to support different architectures, like [ThreadX](https://github.com/azure-rtos/threadx) which has in its CI the build with Cortex-M0/3/4/7.

If we follow this logic, it could be interesting to add in the CI the build of the reference binary with the following buildchain (for example):
- ARM R5 g++ compiler/linker (cortex-R5 baremetal): armr5-none-eabi-g++
- ARM v8 Linux g++ compiler/linker (cortex-A53 linux): aarch64-linux-gnu-g++
- ARM v8 g++ compiler/linker (cortex-A9 baremetal) : arm-none-eabi-g++
- GNU RISC-V Cross C++ compiler/binder (baremetal): riscv64-unknown-elf-g++
- GNU ARM Cross C++ compiler/associator (cortex-M3 baremetal): arm-none-eabi-g++.

> Cortex-A53 and Cortex-R5 are integrated on the [Zynq UltraScale+MPSoC](https://www.xilinx.com/products/silicon-devices/soc/zynq-ultrascale-mpsoc.html) platforms from Xilinx.

> RISC-V is used on the [Polarfire SoC](https://www.microsemi.com/product-directory/soc-fpgas/5498-polarfire-soc-fpga) from Microchip.

Then, for the execution of the unit tests associated with these cross-compiled binaries, they can be performed on emulators like QEMU or Renode.

LeStarch, 2022-04-14:
That is the dream, but first we need solid toolchain implementations for these platforms...

LeStarch, 2022-06-30:
#1443 has another CI improvement.

matt392code, 2024-12-28:
Toolchain implementation for ARM R5 (Cortex-R5) 
Set up the cross-compiler toolchain structure:
[arm-r5-os-support.txt](https://github.com/user-attachments/files/18264390/arm-r5-os-support.txt)
Create the platform-specific configuration for F Prime:
[arm-r5-platform.txt](https://github.com/user-attachments/files/18264391/arm-r5-platform.txt)
Create a basic platform support layer:
[arm-r5-os-support-1.txt](https://github.com/user-attachments/files/18264394/arm-r5-os-support-1.txt)
This implementation provides:
Toolchain Configuration:
- Sets up ARM cross-compiler
- Configures appropriate compiler flags for Cortex-R5
- Enables hardware floating-point unit
- Sets up optimization flags
Platform Configuration:

- Defines platform-specific settings
- Configures memory management
- Sets up bare metal scheduler options
- Defines hardware-specific parameters
OS Support Layer:

- Platform initialization
- Bare metal scheduling
- Memory management
- Task creation and management
- Hardware-specific features

To use this:
Install the required tools:
sudo apt-get install gcc-arm-none-eabi
sudo apt-get install binutils-arm-none-eabi
sudo apt-get install libnewlib-arm-none-eabi
Configure your build:
cmake -DCMAKE_TOOLCHAIN_FILE=toolchain-arm-r5.cmake -DPLATFORM=arm-r5 ..
Build the project:
make

thomas-bc, 2025-06-11:
I'm going to close this issues:
- **lots** of CI improvements since this ticket was open
- lots to come --> label https://github.com/nasa/fprime/labels/CI
- this isn't tracking specific work anymore

closed 2025-06-11 by thomas-bc