nasa/fprime #1388: BufferLogger Unit Test fails in Docker Ubuntu 20.04

Shad84, 2022-04-12:
| | |
|:---|:---|
|**_F´ Version_**| 2.1.0 |
|**_Affected Component_**| Svc/BufferLogger |
---
## Problem Description

Running 
```
fprime-util check 
```
I get the following 
```
The following tests FAILED:
         61 - Svc_BufferLogger_ut_exe (Failed)
```
It seems the problem occurs in BufferLogger/test/ut/Errors.cpp:177

Where ASSERT_EVENTS_SIZE(2) is expecting 2 events to be in the log, but 
```this->eventsSize``` seems to have only 1 at that point in the code.
## How to Reproduce

1. Spin a docker container running Ubuntu 20.04 (focal)
2. run fprime-util check

## Expected Behavior

For the units tests to pass.

LeStarch, 2022-04-13:
We just checked this on devel and an ubuntu 20.04 container.  It appears to be working correctly.  I will close for now, but if an error is detected on F´ v3+ feel free to reopen.

Shad84, 2022-04-21:
Thanks @LeStarch for looking into this, it is unfortunate that you weren't able to reproduce it. It happens for me consistently in docker containers running on a multitude of platforms. (Docker on Windows, Docker on Ubuntu, Docker on MacOS)
However, there has been another report about this issue using v3.0.1:

[Svc/BufferLogger tests](https://github.com/nasa/fprime/issues/509#)
#509

LeStarch, 2022-04-21:
I just reopened #509 and we'll look into it.  A report on v3.0.1 helps. Any chance you can post the docker file you are using?

LeStarch, 2022-05-03:
Turns out, the BufferLogger tests fail when you are root.  These tests are checking to ensure that a write permissions error will be trapped, but root always has write permissions.  The fix is to run these tests as a non-root user.

closed 2022-05-03 by LeStarch