nasa/fprime #1411: error in fprime-util generate

dmkunsman, 2022-04-17:
I have been trying to follow the tutorials to install fprime, but when I try to use the fprime-util to generate a build system I get this error.

-- Searching for F prime modules in: /run/media/root/Storage/fprime
-- Autocoder constants file: /run/media/root/Storage/fprime/config/AcConstants.ini
-- Configuration header directory: /run/media/root/Storage/fprime/config
-- [fpp-tools] /run/media/root/Storage/fpp/compiler/bin/fpp-depend version v1.0.1-5-g78d86ff3 not expected version v1.0.1
CMake Error at /run/media/root/Storage/fprime/cmake/required.cmake:21 (message):
  fpp tools were not found.  Please install the fpp tool suite.
Call Stack (most recent call first):
  /run/media/root/Storage/fprime/cmake/FPrime.cmake:25 (include)
  CMakeLists.txt:32 (include)

-- Configuring incomplete, errors occurred!

I tried installing fpp compiler by source but I still get the same error

LeStarch, 2022-04-19:
To fix this error you should install the FPP tools at version "v1.0.1".  

Alternatively, you can try:

```
pip install setuptools_scm urllib3
pip install fprime-fpp
```
To get version `v1.0.1`.  The second method is the intended installation path for the next release, so if that causes issues, let me know!

LeStarch, 2022-05-03:
Never heard back on this one, however; F´ will ship will a new metapackage to make installing this easier and the tools check version compatibility.  Thus this should not happen again.

closed 2022-05-03 by LeStarch

capsulecorplab, 2022-05-08:
> To fix this error you should install the FPP tools at version "v1.0.1".
>
> Alternatively, you can try:
>
> ```
> pip install setuptools_scm urllib3
> pip install fprime-fpp
> ```
>
> To get version `v1.0.1`. The [...]

I'm getting a `ModuleNotFoundError: No module named 'setuptools_scm'` error with `pip install fprime-fpp`