nasa/fprime #1418: Segfault In AssertFatalAdapter

LeStarch, 2022-04-22:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

The fprintf call to put the assert message into stderr segfaults on linux systems.  Switch back to printf.

LeStarch, 2022-04-22:
Here is the series of events:

1. Ref ships with a stack size per-thread of 16K via `instances.fpp`
2. AssertFatalAdapter calls fprintf(stderr...
3. stderr is unbuffered
4. fprintf tries to buffer stderr by allocating a large buffer on the stack (rumored to be 8K)
5. 8K on an already saturated stack of only 16K causes a seg-fault.

Solution: swtich to stdout and printf, as stdout is a buffered stream and does not trigger an allocation.

vietjtnguyen, 2022-04-22:
I'm curious, would switching to a `write()` syscall to `STDERR_FILENO` resolve this? I would expect that to not generate any additional buffers on the stack, though you'll probably need some logic to handle partial/incomplete writes.

LeStarch, 2022-04-23:
Yes, it does.  As does an `fwrite`.

PR #1427 (2022-04-29): lestarch: fixing issues with small stack sizes
| | |
|:---|:---|
|**_Originating Project/Creator_**| |
|**_Affected Component_**|  |
|**_Affected Architectures(s)_**|  |
|**_Related Issue(s)_**|  |
|**_Has Unit Tests (y/n)_**|  |
|**_Builds Without Errors (y/n)_**|  |
|**_Unit Tests Pass (y/n)_**|  |
|**_Documentation Included (y/n)_**|  |

---
## Change Description

Bump the default stack size for RPI and Ref to 64K as 16K as a default was a bit small.  Removing the fprintf(stderr... to fix #1418
closed 2022-04-29 by LeStarch