nasa/fprime #1431: Unanticipated Regenerations

LeStarch, 2022-05-03:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

Ref occasionally regenerates the build-cache between the build and the installation step.  Why?

Reports of regeneration of AC files if they are touched.  Why?

JackNWhite, 2022-09-28:
I experience this problem on our project, but it always happens - not just sometimes.

Perhaps turning on verbosity might shed some light on this. I ran `VERBOSE=1 make` from my unit-testing build cache and on only the third line:
```
Re-run cmake file: Makefile older than: MyComponent/MyComponentComponentAi.xml
```
I touched the Makefile, and got the same for `CMakeFiles/cmake.check_cache`. Could these files be created after the autocoders have completed?

LeStarch, 2022-09-28:
@JackNWhite this is great insight!  Are you using FPP to model?

If so, then the issue is that the `Ai.xml` files (generated during the build) are still marked as influencing the CMake system...which means it will randomly regenerate.

We need to add a check in the CMake system to only make that dependence if and only if the Ai.xml is not a generated product.

JackNWhite, 2022-09-29:
Yes, it's all FPP now.

PR #1682 (2022-10-03): lestarch: cleaned-up cmake autocoder handling. Fixes: #1431, #1680, and #1681
| | |
|:---|:---|
|**_Originating Project/Creator_**| |
|**_Affected Component_**|  |
|**_Affected Architectures(s)_**|  |
|**_Related Issue(s)_**|  |
|**_Has Unit Tests (y/n)_**|  |
|**_Builds Without Errors (y/n)_**|  |
|**_Unit Tests Pass (y/n)_**|  |
|**_Documentation Included (y/n)_**|  |

---
## Change Description

This PR addresses three inconsistencies in the build system:

1. #1431, caused by attempting to mark the wrong files automatically as CMAKE_CONFIGURE_DEPENDS
2. #1681, caused by attempting to automat [... cut]
commit 4758c631 (2022-10-03): lestarch: cleaned-up cmake autocoder handling. Fixes: #1431, #1680, and #1681 (#1682)

* lestarch: cleaned-up cmake autocoder handling. Fixes: #1431, #1680, and #1681

* lestarch: removing tab to fix spacing

* lestarch: removing debug code

* lestarch: typo in improved docs
closed 2022-10-03 by github-actions[bot]