nasa/fprime #1446: [ImportError: libxslt.so.1: cannot open shared object file: No such file or directory] while building unit tests for raspberrypi

capsulecorplab, 2022-05-11:
| | |
|:---|:---|
|**_F´ Version_**| v3.0.0 |
|**_Affected Component_**| Ref/SignalGen |
---
## Problem Description

Encountered the following traceback error while attempting to build the `Ref/SignalGen` unit tests for raspberry pi
```
(fprime-venv) pi@raspberrypi:~/fprime/Ref/SignalGen$ fprime-util build raspberrypi --ut
[WARNING] Failed to find settings file: /home/pi/fprime/Ref/settings.ini
[ 10%] Built target Fw_Cfg
[ 20%] Built target codegen
[ 20%] Generating DeserialStatusEnumAc.hpp, DeserialStatusEnumAc.cpp
Traceback (most recent call last):
  File "/home/pi/fprime/Autocoders/Python/bin/codegen.py", line 17, in <module>
    from lxml import etree
ImportError: libxslt.so.1: cannot open shared object file: No such file or directory
make[3]: *** [F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/build.make:74: F-Prime/Fw/Types/DeserialStatusEnumAc.hpp] Error 1
make[2]: *** [CMakeFiles/Makefile2:18649: F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:31314: Ref/SignalGen/CMakeFiles/Ref_SignalGen_ut_exe.dir/rule] Error 2
make: *** [Makefile:11736: Ref_SignalGen_ut_exe] Error 2
[ERROR] CMake erred with return code 2

```

## How to Reproduce

On Ubuntu:
1. checkout https://github.com/capsulecorplab/fprime/commit/6f864e0a73f8372cfcc851306bbc96b18ae88b84
2. `cd` into `fprime/Ref/`
3. generate unit tests for raspberrypi, using `fprime-util generate raspberrypi --ut`
4. copy `fprime` directory to raspberrypi

On raspberry pi:
5. `cd` into `fprime/Ref/SignalGen`
6. build unit tests with `fprime-util build raspberrypi --ut`

## Expected Behavior

Should build unit tests

capsulecorplab, 2022-05-11:
It's worth noting I also ran into `/bin/sh: 1: cheetah-compile: command not found` not found error while building, similar to #93, but I symbolically linked my local python installation to `/usr/local/bin/cheetah-compile` as a work around

LeStarch, 2022-05-11:
However, I should also mention most projects just compile and run the UTs on Linux on Ubuntu.  That is usually considered enough for the Unit Testing if the code.  Then projects compile and copy the flight software binary over to the hardware (RPI) and run system tests against that to prove it runs as a system.

capsulecorplab, 2022-05-13:
> If you've generated on Ubuntu, then you must build on Ubuntu. If you want to build on the RPI itself, you must generate there.
> 
> You should be able to run:
> 
> ```
> fprime-util build raspberrypi --ut
> ```
> 
> To cross-compile the UTs on Ubuntu for the RPI. Then copy the binaries over to the RPI and run them there.

Still ran into the same error after building unit test binaries on Ubuntu, but installing `libxslt-dev` with `apt` on the pi seems to have resolved the issue!

capsulecorplab, 2022-05-13:
> However, I should also mention most projects just compile and run the UTs on Linux on Ubuntu. That is usually considered enough for the Unit Testing if the code. Then projects compile and copy the [...]

I figured that's the typical workflow; I just needed a means of verifying whether the cross-compiler was properly building the binaries, without having having to do a system-level test

closed 2022-05-13 by capsulecorplab