nasa/fprime #1455: add/generate `requirements.txt` as a "lockfile" for python packages

capsulecorplab, 2022-05-15:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Add or generate a `requirements.txt`, containing pinned versions for all python packages and dependencies. Such a `requirements.txt` can be generated from `setup.py` using a tool s.a., `pip-compile` from [pip-tools](https://github.com/jazzband/pip-tools/). `requirements.txt` files could also be generated for `fprime-tools`, `fprime-gds`, and/or `fprime-fpp`

## Rationale

A `requirements.txt`, containing pinned versions for all python packages and dependencies, would provide an optional means to ensure reproducible installation of the _working_ versions of all python packages and dependencies, using `pip install -r requirements.txt`, essentially functioning akin to a `package-lock.json` used in NodeJS projects managed with `npm`.

ThibFrgsGmz, 2022-05-15:
Perhaps [Poetry](https://python-poetry.org/) can also be considered?
From memory, this tool makes it easy to manage and track such dependencies.

I made a small demo applied to FPrime here: https://github.com/nasa/fprime/discussions/1005

capsulecorplab, 2022-05-15:
@ThibFrgsGmz The reason I suggested [pip-tools](https://github.com/jazzband/pip-tools/) is because it works with `setup.py`. Does Poetry do the same? Looks interesting, nonetheless

ThibFrgsGmz, 2022-05-15:
@capsulecorplab From what I remember, it handles the setup.py file.
But it's been a while since I've prototyped with this tool, so my memories are a bit hazy.

I see people using this tool in large software production, so I guess it has some advantages.

ThibFrgsGmz, 2022-05-24:
Note: there are other tools similar to Poetry like [Pipenv](https://pypi.org/project/pipenv/) and [PDM](https://pdm.fming.dev/). 
Poetry and PDM would be better than Pipenv in terms of performance. And PDM would be better than Poetry on time cost and compatible dependency resolution.

LeStarch, 2022-06-03:
To expand a bit on what we are thinking here.

##On Packages##

`fprime-tools`, `fprime-gds`, and `fprime-fpp` are python packages. In their development form, and as packages these should have the loosest requirements possible.  This allows PIP to pick the newest packages possible that work together. This pattern works for generic packages, and for developers of these packages.

It also allows an easier update path for packages as there aren't firm, possibly conflicting, package version.

##On Projects##

However, most if not all of our users aren't developers.  They are projects.  F´ is also a project in a similar regard.  These projects use the packages, but need stability and "known working configurations".  In order to provide this, F´ will switch from a "package" (e.g. setup.py) and instead will publish a `requirements.txt` generated to include versions of all packages in the dependency tree (`fprime-gds`, `fprime-tools`, `fprime-fpp`, and dependencies there-in).

Users of F´ will install using the following syntax and will receive a working set configuration for packages that will allow the tools to run as expected.

```
pip install -r fprime/requirements.txt
```

By fixing all versions (automatically) for projects/users tool stability will increase by removing the "churn" of underlying packages. In addition, development will be easier because (for developers) package versions will be less strict.  Lastly, formal projects (e.g. a robot team at a university) could choose to publish their own `requirements.txt` should they have need of specific extra requirements without running afoul of the strict package requirements we are trying to remove.

capsulecorplab, 2022-06-04:
@LeStarch Any particular reason we can't maintain both a `setup.py` and `requirements.txt`? The `install_requires` variable in `setup.py` can still serve as a minimal dependency list (akin to a `package.json` used by npm/yarn) that's then used to generate a `requirements.txt` to serve as the aforementioned "lockfile" (akin to a `package-lock.json`). As mentioned, `pip-tools` comes with a `pip-compile` utility that does exactly that https://github.com/jazzband/pip-tools/#example-usage-for-pip-compile

closed 2022-06-16 by LeStarch