nasa/fprime #1456: [ERROR] Dictionary version 3.0.1 is not in supported range: 0.0.0-3.0.0. Please upgrade fprime-gds.

capsulecorplab, 2022-05-15:
| | |
|:---|:---|
|**_F´ Version_**| v3.0.0 |
|**_Affected Component_**|  |
---
## Problem Description

I'm encountering the following error while attempting to run `fprime-gds` to connect to my raspberry pi running a cross compiled binary for either **Ref** or **RPI**

```
[ERROR] Dictionary version 3.0.1 is not in supported range: 0.0.0-3.0.0. Please upgrade fprime-gds.
[ERROR] Failed to run application: **/fprime/venv/bin/python3 -u -m flask run --host 127.0.0.1 --port 5000. Error: **/fprime/venv/bin/python3 stopped with code 1 sooner than 2 seconds.
```

Here's a printout of `pip freeze` from my python virtual environment
```
aniso8601==9.0.1
argcomplete==1.12.3
arrow==1.2.2
attrs==21.4.0
binaryornot==0.4.4
certifi==2021.10.8
chardet==4.0.0
charset-normalizer==2.0.12
Cheetah3==3.2.6
click==8.1.3
cookiecutter==1.7.2
Flask==1.1.2
Flask-RESTful==0.3.8
fprime-fpp==1.0.1
fprime-gds==3.0.2
fprime-tools==3.0.1
idna==3.3
iniconfig==1.1.1
itsdangerous==2.0.1
Jinja2==2.11.3
jinja2-time==0.2.0
lxml==4.6.3
Markdown==3.3.4
MarkupSafe==1.1.1
packaging==21.3
pep517==0.12.0
pexpect==4.8.0
pip-tools==6.6.1
pluggy==0.13.1
poyo==0.5.0
ptyprocess==0.7.0
py==1.11.0
pyparsing==3.0.9
pytest==6.2.4
python-dateutil==2.8.2
python-slugify==6.1.2
pytz==2022.1
requests==2.27.1
setuptools-scm==6.0.1
six==1.16.0
text-unidecode==1.3
toml==0.10.2
tomli==2.0.1
urllib3==1.26.9
Werkzeug==2.0.1
```

## How to Reproduce

In a python virtual environment:
1. build **Ref** or **RPI** binary for raspberry pi, follow either [Cross Compilation tutorial](https://nasa.github.io/fprime/Tutorials/CrossCompilation/Tutorial.html) or [RPI Demo Tutorial](https://github.com/nasa/fprime/blob/master/RPI/README.md)
2. run `fprime-gds -n --dictionary build-artifacts/raspberrypi/dict/RefTopologyAppDictionary.xml` (for Ref) or `fprime-gds -n --dictionary build-artifacts/raspberrypi/dict/RPITopologyAppDictionary.xml` (for RPI)

## Expected Behavior

Should open ground station desktop UI

Joshua-Anderson, 2022-05-16:
What do you see when you run `git describe --tags --always` in your fprime deployment?

capsulecorplab, 2022-05-16:
> What do you see when you run `git describe --tags --always` in your fprime deployment?

v3.0.0-1-g6f864e0a7

Joshua-Anderson, 2022-05-16:
I wonder if this is a bug in the [fprime framework version detection script](https://github.com/nasa/fprime/blob/devel/Autocoders/Python/src/fprime_ac/utils/version.py). What is the version in your xml dictionary file?

capsulecorplab, 2022-05-16:
> I wonder if this is a bug in the [fprime framework version detection script](https://github.com/nasa/fprime/blob/devel/Autocoders/Python/src/fprime_ac/utils/version.py). What is the version in your xml dictionary file?

`<dictionary topology="RPI" framework_version="3.0.1.dev1+g6f864e0a7">`

Joshua-Anderson, 2022-05-16:
`v3.0.0-1-g6f864e0a7` getting turned into `3.0.1.dev1+g6f864e0a7` doesn't look right. Definitely seems like a versioning script bug.

capsulecorplab, 2022-05-16:
@Joshua-Anderson any suggested work around? I tried changing the version number in the dictionary as a crude fix, but that still yields the same error without the version number

Joshua-Anderson, 2022-05-16:
Just removing `framework_version` or setting it to `framework_version="3.0.0"` should work. Fixing the version script shouldn't be too bad either.

capsulecorplab, 2022-05-16:
@Joshua-Anderson Setting `framework_version="3.0.0"` worked!

capsulecorplab, 2022-05-16:
I'll see if I can take a shot at updating version script some time this weekend then open a PR

LeStarch, 2022-06-30:
We ran into this while trying to release.  This check is now effectively removed.  We need to invent a schema version instead of using the software version.  The GDS should then check the schema version.  Checking sw version directly causes problems like this.

closed 2022-06-30 by LeStarch