nasa/fprime #1465: [ERROR: Port xml specification file Fw/Cmd/CmdPortAi.xml does not exist!] encountered while generating implementations following GPS tutorial

capsulecorplab, 2022-05-19:
| | |
|:---|:---|
|**_F´ Version_**| v3.0.0 |
|**_Affected Component_**| Gps |
---
## Problem Description

Encountered following error while attempting to generate implementations with `fprime-util impl` in the `GpsApp/Gps` directory while following the **Coding Our Module** section of the [F´ GPS Tutorial](https://nasa.github.io/fprime/Tutorials/GpsTutorial/Tutorial.html)

```
[~/fprime/GpsApp/Gps]$ fprime-util impl
[WARNING] Failed to find settings file: /home/kasm-user/fprime/GpsApp/settings.ini
[100%] Built target codegen
[100%] Generating ../../../Gps/GpsComponentImpl.hpp-template, ../../../Gps/GpsComponentImpl.cpp-template
Parsing Component Gps
Reading external dictionary /home/kasm-user/fprime/GpsApp/Gps/Commands.xml
Reading external dictionary /home/kasm-user/fprime/GpsApp/Gps/Telemetry.xml
Reading external dictionary /home/kasm-user/fprime/GpsApp/Gps/Events.xml
ERROR: Port xml specification file Fw/Cmd/CmdPortAi.xml does not exist!
make[3]: *** [GpsApp/Gps/CMakeFiles/GpsApp_Gps_impl.dir/build.make:74: ../Gps/GpsComponentImpl.hpp-template] Error 255
make[2]: *** [CMakeFiles/Makefile2:7331: GpsApp/Gps/CMakeFiles/GpsApp_Gps_impl.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:7338: GpsApp/Gps/CMakeFiles/GpsApp_Gps_impl.dir/rule] Error 2
make: *** [Makefile:2152: GpsApp_Gps_impl] Error 2
[ERROR] CMake erred with return code 2
```

## How to Reproduce

(Optional): run following steps in [kasm-fprime-workspace](https://github.com/capsulecorplab/kasm-fprime-workspace)

1. clone and checkout https://github.com/capsulecorplab/fprime/commit/3fd0d6cdd887492c38a16f2beb40eccc078e8766
2. `cd` into `GpsApp/`
3. generate build cache with `fprime-util generate`
4. `cd` into `Gps/`
5. generate implementations with `fprime-util impl`

## Expected Behavior

Should generate `GpsComponentImpl.cpp-template` and `GpsComponentImpl.hpp-template`

LeStarch, 2022-06-27:
We need to fix or abandon the GPS tutorial.  My thoughts are to abandon it in favor of a single more verbose systems reference....but I'd appreciate input.

capsulecorplab, 2022-06-30:
I haven't tested it with the hardware, but it's worth noting that the `GpsApp` on @danjwait's [fprime fork](https://github.com/danjwait/fprime/tree/GpsApp/GpsApp) seems to build fine. Assuming it works, perhaps we keep the GpsApp project files, but replace the tutorial? I was suggesting an I2C example in #1329.

capsulecorplab, 2022-06-30:
I spoke too soon; @danjwait's GpsApp fork *was* building, but encounters a CMake Error after updating my dev environment with python packages installed from https://github.com/nasa/fprime/blob/v3.1.0/requirements.txt

LeStarch, 2022-07-21:
> ... more in-depth systems reference

This is currently under development!  We expect to have something we can start showing towards end of summer or so.

capsulecorplab, 2022-08-23:
Sorry for the delayed response on my end as well (recently got back from a long roadtrip)~ I updated my local dev environment with python packages as per the requirements.txt in fprime v3.1.1 release and checked out @danjwait's GpsApp branch at https://github.com/danjwait/fprime/commit/96f88973ac2a61e361a5c68e611d37c521b0a314 and was able to rerun `fprime-util generate` without encountering the `ERROR: Port xml specification file Fw/Cmd/CmdPortAi.xml does not exist!` error! However, running `fprime-util impl` in `Gps/` isn't generating a `GpsComponentImpl.cpp-template` and `GpsComponentImpl.hpp-template`. Though, I do see a `Gps.cpp-template` and `Gps.hpp-template`.

LeStarch, 2022-09-26:
Yes, the output names have been reworked for simplicity.  No more `ComponentImpl`

capsulecorplab, 2022-09-26:
Okay~ So theoretically, if I rename `Gps.cpp-template` to `Gps.cpp` and `Gps.hpp-template` to `Gps.hpp-template`, it should compile with `fprime-util build`?

LeStarch, 2022-09-26:
Yes.  You'd need to update the make system too, as that CMakeList.txt points to the old naming.  Keep in mind, these stubs are empty.

LeStarch, 2023-05-03:
The GDS tutorial has been officially abandoned as it was never updated to run with FPP.  Closing this issue.

closed 2023-05-03 by LeStarch