nasa/fprime #1478: fpp-tools version incompatible

juliannerice, 2022-05-25:
When following most up-to-date installation instructions after trying to run `fprime-util generate` under Ref we're receiving the error below.

Steps used to try to solve the problem:
- Ran `pip install --upgrade fprime-tools fprime-gds` to make sure fpp-tools were installed and reasurred they were with `pip install fprime-fpp`
- When running the suggested command in the error, (`pip install -U "/home/julianne/fprime/" `), an error with sbt popped up. Even after making sure sbt was installed and trying to get the fpp-tools from that method, we're still getting the same error.  

The error doesn't allow a successful generation or build after a fresh clone of FPrime.

[image]

LeStarch, 2022-05-27:
Are you running `pip install fprime` or `pip install /path/to/fprime`?

nwasselle, 2022-05-28:
I am running ```pip install fprime/``` in the directory above the fprime directory.

capsulecorplab, 2022-06-03:
Bumping issue, as I'm encountering the same problem

capsulecorplab, 2022-06-03:
I was able to generate and build both **Ref** and **RPI** after cloning https://github.com/fprime-community/fprime-fpp/commit/336fa955a221e2ae322c613a4fb7ff3e6899e343 and running `FPP_TOOLS_VERSION=v1.0.1-84-g7271178b pip install .` in the `fprime-fpp` directory. Here's my [requirements.txt](https://github.com/nasa/fprime/files/8836205/requirements.txt) piped from `pip freeze > requirements.txt` - feel free to use it for #1455

LeStarch, 2022-06-03:
I am working on fixing this issue. Unfortunately it is fairly gnarled-up in: https://github.com/nasa/fprime/issues/1455.  Once we sort out a stable set of requirements and how these are installed, we should be able to ensure that the package installations are stable.

The issue is this: currently the FPP packge needed for devel/fprime requires a non-released version of FPP. This means it needs to run the Scala/JAR files not a native compilation. It also needs to build those as part of the install.  Once we get the requirements sorted, we'll automate the production of those native builds and ensure that we release something more often.

As a work-around, you can try the following:

1. Ensure that Java (Java 11 or newer) is installed an on the path.   `which java` should return a valid path.
2. Ensure you are on the latest `devel`
3. Run `pip install -v -U fprime/` this will attempt to build the package.  The `-U` is important as it will force an upgrade to all packages, even ones already installed.  The `-v` will produce extra errors and help us debug.

I hope to have cleaner requirements sorted next week.

LeStarch, 2022-06-04:
One other requirement occurred to me, please see edited list above.

juliannerice, 2022-06-07:
I just tried reinstalling following the Install.MD instructions and it worked! I used steps 1-3 above and ran `pip install fprime/` without any errors.

LeStarch, 2022-06-27:
As a note, the following is now recommended on the latest `devel`.  We apologize for the churn, but should have gotten things in-order.

```
pip install -r fprime/requirements.txt
```

Closing

closed 2022-06-27 by LeStarch