nasa/fprime #1486: Add standard FPP types

timcanham, 2022-06-02:
| | |
|:---|:---|
|**_F´ Version_**| 3.0 |
|**_Affected Component_**| None |
---
## Feature Description

Add some common types that users could use. Things like:

ON, OFF (ex. a device)
HIGH, LOW (logic level)
OPEN, CLOSED (file, cover, etc)
IN, OUT (direction of flow)
ACTIVE, INACTIVE (behavior)
HEALTHY, SICK, FAILED (Device health)

## Rationale

Would avoid users defining them again and again.

ThibFrgsGmz, 2022-06-02:
And ``INOUT`` in addition for direction of flow ?

LeStarch, 2022-09-26:
I **really** want this.

LeStarch, 2022-09-26:
We should also scan the code base for common types implemented within components and see if we can make them generic.

codeflight1, 2022-09-30:
I'm currently working on this. Which module should the definitions reside in?

LeStarch, 2022-10-20:
This was merged in.  Is it complete?

closed 2022-10-20 by LeStarch