nasa/fprime #1487: Include reference implementation for concepts mentioned in Porting Guide

capsulecorplab, 2022-06-03:
| | |
|:---|:---|
|**_F´ Version_**| v3.0.0 |
|**_Affected Component_**|  |
---
## Feature Description

As a follow-up to https://github.com/nasa/fprime/issues/76, the [Porting Guide](https://nasa.github.io/fprime/UsersGuide/dev/porting-guide.html) should include a reference implementation for the concepts mentioned in the guide, i.e.,

1) CMake Toolchain and F´ platform files
2) Fw::Types and configuration
3) Hardware drivers
4) OSAL support

## Rationale

As to provide a working example of how these concepts should be implemented

ThibFrgsGmz, 2022-06-03:
I support this idea because it will make it easier for stakeholders to estimate the ROM costing associated with porting FPrime to a specific unsupported hardware platform and operating system. This is a major issue on my side to convince our working groups in adopting the framework.

LeStarch, 2022-06-27:
@pelmini look here too.  These are some other ideas on stuff to include.

LeStarch, 2022-09-26:
We need to add this to system reference.

matt392code, 2024-12-27:
# F Prime Reference Implementations

## 1. CMake Toolchain and Platform Files

### Example Toolchain File (arm-cortex-toolchain.cmake)
```cmake
set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_SYSTEM_PROCESSOR arm)

# Specify the cross compiler
set(CMAKE_C_COMPILER    arm-none-eabi-gcc)
set(CMAKE_CXX_COMPILER  arm-none-eabi-g++)
set(CMAKE_ASM_COMPILER  arm-none-eabi-gcc)

# Search for programs only in host directories
set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)

# Compiler flags
set(CMAKE_C_FLAGS   "${CMAKE_C_FLAGS} -mcpu=cortex-m4 -mthumb")
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mcpu=cortex-m4 -mthumb")
```

### Example Platform File (stm32-platform.cmake)
```cmake
include(${CMAKE_CURRENT_LIST_DIR}/arm-cortex-toolchain.cmake)

# Platform-specific definitions
add_definitions(-DSTM32F4)
add_definitions(-DHSE_VALUE=8000000)

# Platform-specific compiler flags
set(PLATFORM_C_FLAGS
    "-mcpu=cortex-m4 \
     -mfloat-abi=hard \
     -mfpu=fpv4-sp-d16"
)

set(PLATFORM_CXX_FLAGS "${PLATFORM_C_FLAGS}")

# Link flags including startup and linker script
set(PLATFORM_LINKER_FLAGS
    "-T${CMAKE_SOURCE_DIR}/platform/stm32/linker/STM32F407VGTx_FLASH.ld \
     -mcpu=cortex-m4 \
     -mfloat-abi=hard \
     -mfpu=fpv4-sp-d16 \
     -specs=nano.specs"
)
```

## 2. Fw::Types Configuration

### Example Types Configuration (FpConfig.hpp)
```cpp
#ifndef FP_CONFIG_HPP
#define FP_CONFIG_HPP

// Include platform-specific definitions
#include <PlatformTypes.hpp>

namespace Fw {

    // Integer types
    typedef PlatformU8    U8;
    typedef PlatformI8    I8;
    typedef PlatformU16   U16;
    typedef PlatformI16   I16;
    typedef PlatformU32   U32;
    typedef PlatformI32   I32;
[... 257 lines cut ...]

        return QUEUE_EMPTY;
    }

    Queue::~Queue() {
        if (m_handle != nullptr) {
            vQueueDelete(m_handle);
        }
    }
}
```

These implementations demonstrate:
1. Platform configuration through CMake
2. Type system configuration
3. Hardware driver implementation example
4. OSAL implementations for tasks and queues

Key aspects include:
- Cross-compilation setup
- Platform-specific configurations
- Hardware abstraction
- RTOS integration
- Error handling
- Resource management