nasa/fprime #1488: Python support: drop v3.6 and add v3.10

ThibFrgsGmz, 2022-06-03:
| | |
|:---|:---|
|**_F´ Version_**| 3.0 |
|**_Affected Component_**| python codebase  |
---
## Problem Description

As mentioned by Starch in https://github.com/fprime-community/fprime-tools/pull/53#issuecomment-1111244597, FPrime must support active versions of Python. 

According to https://endoflife.date/python, the Python ``3.6`` version is deprecated as of ``December 23, 2021``, the framework will no longer receive updates or bug fixes.

So theoretically, support for Python ``3.6`` must be removed from the framework (CI runners, Pypi,...).

In addition, Python ``3.10`` is available and active as of ``October 4, 2021`` and therefore should be supported by the framework.

## How to Reproduce

void

## Expected Behavior

- Drop v3.6 support (remove from CI runner, Pypi)
- Add v3.10 support (add CI runner, Pypi)

LeStarch, 2022-06-03:
Agreed.  Don't change this in `fprime` as we'll (likely) removing the `setup.py` in favor of a `requirements.txt`.  We should update this in the `fprime-tools` and `fprime-gds` packages.

ThibFrgsGmz, 2022-06-16:
Side note: the last update of the python version for FPrime was #514

LeStarch, 2022-06-16:
I fixed this in `fprime-tools` and `fprime-gds` when pinning the versions. I bumped the minimum python version and added 3.10 to CI

LeStarch, 2022-06-27:
This is done.

closed 2022-06-27 by LeStarch