nasa/fprime #1493: fprime-util generate produces error in instances.fpp

mrb0y, 2022-06-13:
The following error occurs for me after the entering the fprime-util generate command for the Ref example...

-- Generating FPP location index
fpp-locate-defs
/home/***/fprime/Ref/Top/instances.fpp: 231.5
    type "Drv::TcpClient" \
    ^
error: } expected
CMake Error at /home/***/fprime/cmake/target/fpp_locs.cmake:56 (message):
-- Configuring incomplete, errors occurred!
  Failed to run fpp-locate-deps

I had an adventure installing FPP for WSL2 Ubuntu 20.04 but I think it's properly installed but possibly the root of the problem.

LeStarch, 2022-06-22:
This looks like a possible version of FPP vs version of F´ miss-match.  If you are on the latest devel:

```
pip install -r fprime/requirements.txt
```

You can rerun that above command when you update devel to continue and  track tool dependencies.

LeStarch, 2022-06-27:
Did the above suggestion work?  I am going to close, feel free to reopen if it isn't resolved.

closed 2022-06-27 by LeStarch