nasa/fprime #1506: Show what version of FPRIME is in the branch from the README

sfregoso, 2022-06-16:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

A quick description of the requested feature.

I would be great that at the top of the README it would state what version of FPRIME I would be checking out from this branch, especially the "devel" branch.  If its there, it was not quickly visible to me.

## Rationale

A rationale for why the feature should be implemented in F´.

ThibFrgsGmz, 2022-06-17:
You would like to have a badge that automatically displays the last tagged version ?
GitHub already handles that in the project metadata.
Otherwise these infos are handled by git.
VSCode offers awesome git extension to navigate/visualize for branches.

astroesteban, 2022-06-17:
Yeah I think a locally installed tool like Git Graph or GitLens would be better. I personally use the latter.
The `devel` branch is a moving target and the latest development "version" might become stale rather quickly. The `devel` branch is the latest and greatest so I assume it's whatever the current release version is plus some improvements. Thoughts?

pouriya73, 2022-06-23:
https://github.com/pouriya73/Camputer-Quantum-Programming

LeStarch, 2022-06-27:
You could also run:

```
git describe --tag --always
```

and this will print the version.

LeStarch, 2022-09-30:
I think we can close this issue.  The latest release should be easily available to the right of the window.

closed 2022-09-30 by LeStarch