nasa/fprime #1529: Create a GitHub template `fprime-starter-project`

ThibFrgsGmz, 2022-06-27:
| | |
|:---|:---|
|**_F´ Version_**| none |
|**_Affected Component_**| none |
---
## Feature Description

It would be interesting to create a boilerplate project with "GitHub Template": `fprime-starter-project`.

This project would follow best practices for rapid implementation with F' in a minimalist deployment.

The idea would be to have a project as empty as possible where main() does nothing (or increments a counter).

In terms of philosophy, this would be the same as the following projects:

- [cpp_starter_project](https://github.com/cpp-best-practices/gui_starter_template)
- [vscode_ros2_workspace](https://github.com/athackst/vscode_ros2_workspace).

## Rationale

1. This takes advantage of the ["Template"](https://docs.github.com/en/repositories/creating-and-managing-repositories/creating-a-repository-from-a-template) feature of GitHub.

From this GitHub template, anyone can start with it to build their own deployment little by little.

Implementation tutorials could start from it and the trainer could add features little by little.

2. There are several ways to implement F' :

- The [fprime-sphinx] method (https://github.com/fprime-community/fprime-sphinx)
- The [fprime-arduino] method (https://github.com/fprime-community/fprime-arduino)

Where the OS/Drv/... layers are either in the same folder or in separate repositories. It is not obvious which layout is better.

This would also serve as a reference for the [Setup Deployment](https://github.com/nasa/fprime/blob/devel/docs/UsersGuide/best/development-practice.md#setup-deployment) phase of the reference development practices.

3. When a project starts, the user logically starts from `Ref` and empties it, then adds his own components to be more in control of his topology.

LeStarch, 2022-06-27:
@pelmini take a look here.  I think your "emptied out Ref" goal is a good starting place, and we can adjust from there. 

@ThibFrgsGmz we'll look into this!

capsulecorplab, 2022-10-27:
I've yet to test it, but @SterlingPeet also made a cookiecutter template for generating a project scaffold https://github.com/SterlingPeet/cookiecutter-fprime-component

SterlingPeet, 2022-10-27:
That one is the component, there is a companion deployment template.  I have not created a full GitHub project template, but that’s actually a good idea that I should consider.

    By the way, both of my templates are for F Prime 2.0, more specifically pre-FPP.  I have those updates on the drawing board but not started yet.

Sterling Peet

thomas-bc, 2023-03-16:
FYI - we plan on adding a new `fprime-util new --deployment` and `fprime-util new --project` that will accomplish, I think, the same goal as what this issue was trying to accomplish. Allowing users to very quickly and easily generate a blank deployment, or even project (with `fprime/` as and submodule and everything)
I will start work on that very soon, will be tracked in https://github.com/nasa/fprime/issues/1926

closed 2023-04-17 by thomas-bc