nasa/fprime #1534: Found a possible security concern

JamieSlome, 2022-06-28:
Hello 👋

I run a security community that finds and fixes vulnerabilities in OSS. A researcher (@7h3h4ckv157) has found a potential issue, which I would be eager to share with you.

Could you add a `SECURITY.md` file with an e-mail address for me to send further details to? GitHub [recommends](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) a security policy to ensure issues are responsibly disclosed, and it would help direct researchers in the future.

Looking forward to hearing from you 👍

(cc @huntr-helper)

LeStarch, 2022-06-30:
We don't have a security document setup yet, but I will add it to our list of priority issues. If you'd like to contact me directly about the issue feel free.

JamieSlome, 2022-06-30:
@LeStarch - thanks for your speed and diligence here.

Let me know if you require any support on the security document, as we have helped projects/repositories set them up in the past.

In the meantime, you can view the report directly here:
https://huntr.dev/bounties/b76add26-5680-4d51-8e92-effa36a69597/

It is private and only accessible to maintainers with repository write permissions 👍

LeStarch, 2022-09-26:
In short, we need to bump the cookiecutter version.  This is listed in the "fix fprime-new" PR so I will link and close this issue.

closed 2022-09-26 by LeStarch

7h3h4ckv157, 2022-11-27:
@LeStarch @Joshua-Anderson 

I see that you fixed the issue.
If you don't mind, please mark my report as valid, I'll be a favor for me

Link: `https://huntr.dev/bounties/b76add26-5680-4d51-8e92-effa36a69597/`

psmoros, 2022-12-02:
Hey @7h3h4ckv157 I can mark it as valid and fixed for you if you show me where it's been fixed

7h3h4ckv157, 2022-12-02:
Sorry for the typo! @psmoros  

(After resolving the case --> bump the cookiecutter version )

If you don't mind, please mark my report as valid, It'll be a favor for me.

Link: https://huntr.dev/bounties/b76add26-5680-4d51-8e92-effa36a69597/

7h3h4ckv157, 2023-04-22:
@psmoros Can you proceed now, brother ??