nasa/fprime #1536: (Re)introduce the MBSE methodology in F´

ThibFrgsGmz, 2022-06-28:
| | |
|:---|:---|
|**_F´ Version_**| 4.x |
|**_Affected Component_**| Topology Design/Assembly |
---
## Feature Description

It would be great to be able to perform the ["High Level Design"](https://github.com/nasa/fprime/blob/devel/docs/UsersGuide/best/development-practice.md#high-level-design) step of F' by a graphical tool inspired by the MBSE methodology, or even to fully reintroduce the MBSE methodology in F'.

## Rationale

Currently, the topology is visualized once it has been assembled in the ["Assemble Topology"](https://github.com/nasa/fprime/blob/devel/docs/UsersGuide/best/development-practice.md#assemble-topology) step with the [FPL/FPV](https://github.com/nasa/fprime/blob/devel/docs/Tutorials/MathComponent/Tutorial.md#64-visualizing-the-ref-topology) tools.

Therefore, this does not allow all stakeholders to have a good idea of the overall topology of the software system and to see potential design flaws (not to mention the fact that some system engineers do not want to write lines of code and learn a new language).

On the EU/France side, the [MBSE Eclipse Capella](https://www.eclipse.org/capella/) tool is becoming more and more popular for creating a software system using the MBSE methodology in the space domain.

Maybe there are elements to analyze to introduce them in the framework?

There is also the [RAFCON](https://github.com/DLR-RM/RAFCON) framework from DLR that might be interesting to look at briefly.

LeStarch, 2022-06-28:
@bocchino I think this issue is one for you to think on.

bocchino, 2022-06-28:
You can use the FPL/FPV tools to sketch the design before writing an FPP model. This solution is lightweight and is available now. One could also generate FPP From an MBSE model, but this is future work.

To sketch the design, you can specify a topology directly in the FPL text input format, demonstrated here:

https://github.com/fprime-community/fprime-layout/blob/main/examples/ref/RateGroups/RateGroups.txt

Then you can run the sketch through the tools to generate pictures.

capsulecorplab, 2022-06-30:
Might I also suggest considering [OML](http://www.opencaesar.io/oml/) as a modeling language of choice? It's text-based (i.e., can more easily be version controlled with git), is open-source, and is maintained by JPL.

capsulecorplab, 2022-06-30:
I'm currently going through the [OML tutorial](https://github.com/modelware/oml-tutorial). I can imagine a workflow where an ontology for FPP is defined in OML, which can then queried as json and transpiled to PlantUML (for generating UML diagrams) or FPP, using a template language.

capsulecorplab, 2022-06-30:
I'm currently going through the [OML tutorial](https://github.com/modelware/oml-tutorial). I can imagine a workflow where an ontology for FPP is defined in OML, which can then be queried as json and transpiled to PlantUML (for generating UML diagrams) or FPP, using a template language.

ThibFrgsGmz, 2022-07-02:
An interesting conference on the subject: [Diagrams as Code 2.0 - Simon Brown - GOTO 2021](https://www.youtube.com/watch?v=Za1-v4Zkq5E&ab_channel=GOTOConferences).

Here is the absract of the conference:

"Diagrams as code are becoming a popular way to diagram software architecture, especially for high-level, long-lived documentation. You write the diagram source in a text-based domain-specific language (e.g., PlantUML or Mermaid) or in a programming language, and render the diagrams using web tools or command lines.

The advantages are well understood: writing the diagram source as text allows for easy integration into software development practices and tool chains, and the automatic layout features allow authors to focus on the content.

The problem with this approach is that it is easy for diagrams to get out of sync. This is where the concept of "diagrams as code 2.0" comes in, which allows us to define a software architecture model and the views we want to see, resulting in a consistent set of diagrams generated for us. [...]"

ThibFrgsGmz, 2022-07-02:
Thanks for the advice @bocchino, I'll take a look.

So what I describe is consistent with what you say about "generating FPP models with an MBSE tool".
If it sticks to macroscopic visualization of the topology, that should do the trick.

In my experience, I don't think this type of tool can be relevant when you want to specify implementation details,

capsulecorplab, 2022-07-12:
@bocchino is there documentation on how to setup/run the FPL/FPV tools?

bocchino, 2022-07-12:
The README files explain how to install and use the tools.

https://github.com/fprime-community/fprime-layout
https://github.com/fprime-community/fprime-visual

thomas-bc, 2023-03-08:
Another way to go about this could be to generate the FPP files out of a MBSE model. I have worked on https://github.com/Open-MBEE/Comodo which takes in a SysML model and generates different types of textual representation (or code) out of it. Comodo is neat in that it allows to add your own rules for generating things fairly easily. Adding an "FPP" target is definitely doable with some work.

thomas-bc, 2024-06-04:
I am going to close this at there does not seem to be action items at this time. If more discussion is needed, let's open up a Discussion!

For posterity: `fprime-util visualize` is a quick shortcut to visualize a topology.

closed 2024-06-04 by thomas-bc