nasa/fprime #1562: missing import in run_deployment.py

shalevm, 2022-07-12:
| | |
|:---|:---|
|**_F´ Version_**|3.1.0 |
|**_Affected Component_**| fprime_gds  |
---
## Problem Description

following the `INSTALL.md` tutorial, when testing F´ GDS Installation Via Running HTML GUI, namely:
`fprime-gds -g html -r <path to fprime checkout>/Ref/build-artifacts`

i got a python error from line 23 in the run_deployment.py file:
`root = Path(args.root_dir)`

that indicated that `Path` was unknown,
and indeed, the relevant import was missing.
adding the line
`from pathlib import Path`
fix it.

## How to Reproduce

1. see above

## Expected Behavior

no errors should occur...

rmelick-muon, 2022-07-13:
I have just run into the same issue, I think it is probably caused by this commit to the fprime-gds: https://github.com/fprime-community/fprime-gds/commit/97f549285d6311bb2f75b6735f66da25bf4453f4

Anirban166, 2022-07-19:
Not adding anything new, but I just wanted to emphasize that this is indeed quickly reproducible and comes as the first issue when following the instructions in [`INSTALL.md`](https://github.com/nasa/fprime/blob/master/docs/INSTALL.md#checking-your-f-installation):
```sh
(fprime-venv) <userName>@<systemName> Ref % fprime-gds -g html -r ./build-artifacts
Traceback (most recent call last):
  File "/Users/<userName>/fprime-venv/bin/fprime-gds", line 8, in <module>
    sys.exit(main())
  File "/Users/<userName>/fprime-venv/lib/python3.9/site-packages/fprime_gds/executables/run_deployment.py", line 277, in main
    settings = vars(get_settings())
  File "/Users/<userName>/fprime-venv/lib/python3.9/site-packages/fprime_gds/executables/run_deployment.py", line 23, in get_settings
    root = Path(args.root_dir)
NameError: name 'Path' is not defined
```
It would be nice to bring back the required imports (`pathlib.Path` and `sys`) and add/revert some changes (I just tested the code from the [PR](https://github.com/fprime-community/fprime-gds/pull/87/files) that Russell made and it works/lgtm!) soon so that new fprime users don't have a problem with the initial setup or verification of the GDS installation.

shalevm, 2022-08-16:
Thanks Michael.

closed 2022-08-16 by shalevm