nasa/fprime #1573: Add in MermaidJS Rendering to Docs

LeStarch, 2022-07-19:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

We need to add jekyll-spaceship plugin to our plugins to render mermaidJS

ThibFrgsGmz, 2022-07-20:
This issue needs to be resolved before I continue to convert the documentation to mermaid?

ThibFrgsGmz, 2022-07-20:
The Spaceship's author suggests to use its GitHub Actions https://github.com/jeffreytse/jekyll-deploy-action to use in GitHub Pages.

LeStarch, 2022-09-26:
We should get this up and running in documentation/CI quickly.

matt392code, 2025-01-03:
[fprime-component.mermaid.txt](https://github.com/user-attachments/files/18304397/fprime-component.mermaid.txt)
Thoughts on the value of adding MermaidJS support:

1. Enhanced Documentation Clarity
- MermaidJS would allow creating clear, maintainable diagrams for:
  - Component architectures
  - Data flows
  - State machines
  - Sequence diagrams
  - Deployment topologies

2. Documentation Maintainability
- Diagrams as code means they can be:
  - Version controlled alongside documentation
  - Reviewed in PRs
  - Updated easily
  - Generated programmatically if needed

3. Implementation Path
- Jekyll-spaceship plugin via GitHub Actions seems straightforward
- The jekyll-deploy-action suggested by ThibFrgsGmz looks like a solid approach
- This could be tested in a documentation branch first

4. Benefits for Contributors
- Lower barrier to contributing documentation
- Consistent diagram styling across docs
- No need for external diagram tools
- Easier to maintain and update diagrams

thomas-bc, 2025-01-15:
Done in the above PR!

closed 2025-01-15 by thomas-bc