nasa/fprime #1576: Delete locs.fpp and update-locs files within fprime

legounicycler, 2022-07-20:
| | |
|:---|:---|
|**_F´ Version_**| 3.0.0+ |
|**_Affected Component_**| N/A |
---
## Problem Description

Within fprime, there are a few files sprinked throughout that are not being used anymore. All the files named `locs.fpp` which contain fpp locations for symbols within each build module are no longer needed since there is now a scan that generates a location index in the build system itself. Similarly, the `update-locs` scripts next to these `locs.fpp` files can be deleted.

## How to Reproduce

See the files in the `config/`, `Drv/`, `Fpp/`, `Fw/`, `Ref/`, and `Svc/` directories

## Expected Behavior

This isn't a breaking bug, but it is confusing enough that developers who are using Ref as a reference deployment have tried to use these scripts and files and been confused by why they aren't doing anything (see the discussion here: https://github.com/nasa/fprime/discussions/1555)

LeStarch, 2022-09-26:
We absolutely need to clean this up.

LeStarch, 2022-09-26:
This should be performed as written.  Delete `locs.fpp` and `upadate-locs` files.

closed 2022-10-06 by LeStarch