nasa/fprime #1595: Add the TRL scale associated with the mastery of F´

ThibFrgsGmz, 2022-07-26:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

It would be great to have a TRL scale of key framework features, something like this:

| | |
|:---|:---|
|**TRL 9** | Custom Autocoder |
|**TRL 8** | Hub Pattern for TSP architecture|
|**TRL 7** | Full control of memory management |
|**TRL 6** | Custom serialization protocol |
|**TRL 5** | Ability to write a custom integration test suite |
|**TRL 4** | Adaptation of OSAL to exotic OS |
|**TRL 3** | ... |
|**TRL 2** | ... |
|**TRL 1** | Math Tutorial |

We could perhaps add next to TRLs some of the more common words encountered in skill assessment, such as unaware, aware, informed, capable, recognized, expert, guru.

## Rationale

This will allow stakeholders to

- assess their maturity, their degree of competence with respect to the framework, and therefore the risk of taking this framework for a more or less complicated project
- to set a learning objective to motivate and guide their learning.

LeStarch, 2022-09-26:
I'd be happy to add something like this into our documentation!

matt392code, 2025-01-03:
[fprime-trl.mermaid.txt](https://github.com/user-attachments/files/18304444/fprime-trl.mermaid.txt)
This is an interesting proposal that adapts NASA's TRL concept (typically used for technology maturity) to measure developer proficiency with F Prime. 
Here's a suggested structure that builds on the original concept:
1. Core Competency Levels
TRL 9 (Expert/Guru):
- Custom Autocoder development
- Framework architecture modifications
- Contributing core features

TRL 8 (Advanced):
- Hub Pattern TSP implementation
- Complex deployment topologies
- Performance optimization

TRL 7 (Proficient):
- Memory management customization
- Resource management
- Component optimization

TRL 6 (Competent):
- Custom serialization protocols
- Command/telemetry interfaces
- Error handling patterns

TRL 5 (Intermediate):
- Integration test development
- Component integration
- System validation

TRL 4 (Basic):
- OSAL adaptation
- Basic component development
- Deployment configuration

TRL 3 (Beginner):
- Component implementation
- Basic unit testing
- Framework configuration

TRL 2 (Aware):
- Basic topology understanding
- Component usage
- Build system familiarity

TRL 1 (Novice):
- Math tutorials
- Basic concepts
- Development environment setup

2. Assessment Criteria
Each level could include:
- Required knowledge
- Demonstrable skills
- Example projects
- Common challenges
- Learning resources

3. Learning Pathways
- Structured progression paths
- Prerequisite relationships
- Recommended learning order
- Practical exercises

4. Documentation Integration
- Link to relevant documentation
- Example code for each level
- Best practices
- Common pitfalls

This framework could help:
- Teams assess their F Prime capabilities
- Individuals plan learning paths
- Projects evaluate technical risk
- Organizations plan training

This expanded framework:
1. Provides clear progression paths
2. Maps to common project needs
3. Helps with both assessment and learning
4. Could be integrated with existing documentation

matt392code, 2025-01-03:
[image]

matt392code, 2025-01-03:
[image]

closed 2026-04-15 by thomas-bc