nasa/fprime #1617: Refactor README.md

LeStarch, 2022-08-16:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

Users are discovering F´ via the github repo directly (as opposed to the documentation landing page).  This is causing difficulty in finding and running the installation instructions.

We should look into refactoring the README so it has a similar flow (or links to) our documentation.

LeStarch, 2022-08-16:
@gjwatney feel free to provide specific thoughts here, or direct them to me directly.

LeStarch, 2022-09-26:
In discussing this further, the `README.md` page here should highlight a link to the documentation `Home` and `Getting Started` pages. It should otherwise be minimal.   To be precise:

1. Short intro paragraph (what is F´). Must be succinct.
2. Links to `Documentation Home` and `Getting Started`
3. Links to release notes

saminarp, 2022-10-11:
Hi, 
I would really like to contribute to this as part of Hacktoberfest! I just started out in open-source development and want to help as much as possible. Please let me know how to get started and I will put my best into it. :)
Thank you!!

thomas-bc, 2023-04-24:
Released! Go check out our new README https://github.com/nasa/fprime

closed 2023-04-24 by thomas-bc

ThibFrgsGmz, 2023-04-24:
Yes awesome !