nasa/fprime #1626: Validate that Events can fit within a ComBuffer

Joshua-Anderson, 2022-08-18:
## Feature Description

The autocoder should warn when defining events larger than can be serialized into an event buffer.
Right now if a user defines an event larger than can into a log buffer, F' will assert at runtime during serialization.

## Rationale

Similar to telemetry packetization, F' should prevent definition of impossible to serialize events. This is particularly dangerous for events using strings, since they could successfully serialize in some cases but assert if the strings were too long.

FPP could perform this check, but it will likely require moving the combuffer size definition (which typically drives the size of the logbuffer) from a definition in FpConfig.hpp to FPP so it is available at the autocoding step.

matt392code, 2024-12-28:
[event-buffer-validation.txt](https://github.com/user-attachments/files/18265750/event-buffer-validation.txt)
This proposed solution addresses several key aspects:

1. Compile-time Validation
- Moves buffer size checking from runtime to compile time
- Uses static_assert to catch oversized events during compilation
- Provides clear error messages about size violations

2. Configuration Management
- Defines buffer constants in a central location
- Accounts for header overhead in size calculations
- Makes constants available to FPP autocoder

3. String Handling
- Special validation for string members
- Could be extended to handle variable-length strings
- Prevents runtime surprises with string serialization

4. Integration with FPP
- Structure allows FPP to access buffer size constraints
- Could be extended to generate size checks in autocoded files
- Maintains compatibility with existing F Prime components

bocchino, 2025-08-12:
Duplicate of nasa/fpp#679.

closed 2025-08-12 by bocchino