nasa/fprime #1654: GDS doesnt show file downlink inside GUI

EbenezerA99, 2022-09-14:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

During the workshop at Georgia Tech, I tried to downlink a file using the System Reference Deployment and was not able to see the file in the GUI even though I could access the file by going to the tmp/ directory on the docker machine. I was running the GDS through the rancher desktop setup as specified in the System Reference repo running on a Mac OS Host machine. 

## How to Reproduce

1. Compile the System Reference deployment for the raspberry pi on rancher desktop
2. Upload the file to the raspi
3. Start the GDS on rancher desktop
4. Create a text file using the File Manager component shell command (touch test.txt) through the GDS
5. Attempt to downlink the file using the File Downlink component through the GDS

## Expected Behavior

The file does not show up in the downlink tab of the GDS, but does appear in the tmp/ directory. I'm not sure if it was just an environment problem, but wanted to put up an issue.

thomas-bc, 2023-02-10:
I was able to reproduce the issue with latest devel of fprime-gds. I'll look into it.

LeStarch, 2023-02-10:
This was fixed in fprime-gds/devel

closed 2023-02-10 by LeStarch