nasa/fprime #1673: Dry Run flag for ``fprime-util``

LeStarch, 2022-09-26:
### Discussed in https://github.com/nasa/fprime/discussions/1664

<div type='discussions-op-text'>

<sup>Originally posted by **SterlingPeet** September 21, 2022</sup>
It would be really nice to have a ``--dry-run`` flag in the ``fprime-util`` tool that lists out what steps it is going to do, without actually doing it.  This would be especially convenient for people who want to e.g.: scrape the ``cmake`` flags for an IDE build configuration.

Thanks,

Sterling Peet</div>

LeStarch, 2022-09-26:
Rough implementation:
1. Add `--dry-run` flag
3. Pass dry run value to builder
4. Switch cmake call to print cmake call instead in case of dry run

**Note:** come care for non builder items (purge, coverage) should be take and tested.

LeStarch, 2022-11-30:
@SterlingPeet now that CMake can load settings.ini natively and thus IDEs and straight-up load CMakeLists.txt without additional arguments to work, is this needed?

SterlingPeet, 2022-12-06:
I think the original intent is still desired, for debugging purposes if nothing else.  It would be preferable to have a nice window into how the ``settings.ini`` got compiled into a ``cmake`` call.

LeStarch, 2022-12-06:
@SterlingPeet point taken.  It is good for visibility to see how the command lines are generated regardless of how complicated/simplified they become.

matt392code, 2025-01-03:
[fprime-dry-run.py.txt](https://github.com/user-attachments/files/18304526/fprime-dry-run.py.txt)
Proposed implementation for the dry-run functionality in fprime-util. 
Here are the key features:
1. Added `--dry-run` flag through argparse
2. Created a `DryRunBuilder` class that extends the base `Builder`
3. Overridden `_run_cmake` and `_run_make` to show commands without executing them
4. Added support for showing environment variables and settings
5. Implemented dry-run handling for core commands (generate, build, purge)

Key benefits:
- Shows complete command lines that would be executed
- Displays environment variables and settings
- Preserves the original builder functionality when not in dry-run mode
- Handles error cases gracefully

Example usage:
```bash
fprime-util generate --dry-run
fprime-util build --dry-run
fprime-util purge --dry-run
```

The implementation follows the discussion points from the ticket:
- Provides visibility into how settings.ini is compiled into cmake calls
- Useful for debugging purposes
- Shows command line generation regardless of complexity